canopen-python/canopen #125: Concurrent SDO access casuses an SdoCommunicationError

comkieffer, 2018-11-12:
In a multi-threaded scenario if two threads try to access SDOs at the same time then on of them will fail with an `SdoCommunicationError` after reading a response for the other SDO. 

Is it your goal to make the library thread-safe or should thread synchronization be pushed into the calling code ? If you would like the library to be thread safe I can look into a patch for this problem.

christiansandberg, 2018-11-12:
It wouldn’t hurt to make it thread safe. If you’re up to it you are welcome to provide a patch. Perhaps a simple lock would suffice?

christiansandberg, 2018-11-27:
I would probably add it under the upload and download methods. It won’t work when using the file API, but it will be easier to implement.

comkieffer, 2018-12-06:
This won't be a good enough solution unfortunately. PDOs also seem to interfere with SDOs. I need to test this a bit more but won't have time for at least another week. 

A better solution might to make the SDO code more tolerant to unexpected responses. Instead of assuming that the first message to come back is the response it should wait for so milliseconds and discard any invalid messages it receives in the meantime.

af-silva, 2019-10-16:
Improvement registered and follow up at #143 
Closing this issue.

closed 2019-10-16 by af-silva