canopen-python/canopen #127: Local vs Remote Node

Jmeyer1292, 2018-11-21:
Greetings,

I apologize for what might be a basic question: I'm trying to get this library running with some Curtis controllers and I've been going through a bit of a learning experience as I'm new to CANopen.

So:
 1. In this library, what is the difference between a remote node and a local node? The front page example includes a remote node. The docs show local node. The definitions in the docs themselves are recursive: "A remote note is a remote node".

 2. I created a network, can see my controller, can turn it on and off, and can even read (via SDO) the status of the interlock. I can't seem to follow any of the PDO mapping examples: I get an error telling me the object I want is not in the object dictionary. This seems to happen no matter what I try to map. I have set `curtis_hack = True` prior to calling any of the mapping functions on the node in question.

Thank you!

acolomb, 2018-11-21:
Hi there!

A remote node is the controller you want to talk to, which is not your PC where python-canopen is running. A local node would be this library emulating all the state and behavior just as any other CANopen device would behave. So others can treat your program like you see the Curtis controller.

Do you have a proper object dictionary loaded from an EDS file? For the programmable controllers it is output during compilation by the WinVCL application. For the others you can write it yourself or request one from Curtis support. What model are you using? 

The `curtis_hack` quirk is only applicable to some models that have a broken CANopen implementation. Namely I tested it with the 123xE series. Other controllers (in my case 1222 steering controller) don't need it, but have other shortcomings.

On that one you have to slow down the rapid pace of successive SDO communications that happens when mapping PDOs. Otherwise the controller just drops some commands. There is a delay constant somewhere in the SDO code for this purpose.

Let me know if you have more troubles, I might be able to give you some example code based on my experience with Curtis.

acolomb, 2018-11-21:
And what parameters are you trying to map specifically? Make sure they are contained in the EDS file.

Jmeyer1292, 2018-11-21:
@acolomb Thank you for the prompt response. I have several different controllers at the moment, including both a 123XE and a 1222. I'm currently only trying to communicate with the 123XE.

I haven't touched the onboard software at all: I was given a generic EDS file and the parameter I'm trying to read via PDO is:
```
[3207]
ParameterName=Motor Speed
ObjectType=0x7
DataType=0x0003
AccessType=ro
PDOMapping=1
LowLimit=-32768
HighLimit=32767
ObjFlags=0x0
```

Jmeyer1292, 2018-11-21:
A follow up: I have had some limited success mapping parameters when the `event_timer` field of the PDO is disabled. This seems to work somewhat reliably on the 1222 controller and only semi well on the 123XE. I no longer get no-object errors, but it just won't map some parameters. I suspect a bad EDS.

Another side note: What is the proper procedure for mapping a PDO that is not currently active? In other words, if I read the TPDO status and see that there are two active buffers (TPDO[1] and TPDO[2]) I now know how to change them. How does one add TPDO[3]?

acolomb, 2018-11-22:
The 123xE series only supports two CANopen PDOs out of the box, see section 20.2 of the "VCL Common Functions" documentation. You can implement more, but not via standard CANopen mechanisms, but you will have to manually configure extra "CAN mailboxes" in VCL code.

Regarding the `event_timer`, those controllers (not sure about the 1222) only support asynchronous transmissions.  They will send a TPDO1 just after receiving an RPDO1 for example. Trying to change that behaviour via CANopen will fail or have no effect. Especially configuring a different COB ID will only work from within the VCL code. I tried that to implement some cross-communication among the controllers.

Regarding the EDS, my entry looks like this and mapping the object via the `Motor_RPM` name works:

```
[3207]
ParameterName=Motor_RPM
ObjectType=0x7
DataType=0x0003
AccessType=ro
LowLimit=-32768
HighLimit=32767
PDOMapping=1
```

I just noticed that the EDS file from WinVCL incorrectly has `PDOMapping=0` in there, but that should not matter for python-canopen. You may have to actually upload a program to the 123xE controller, together with a suitable operating system file from Curtis. That procedure will output the (mostly correct) EDS file matching the OS.

acolomb, 2018-11-22:
Oh and by the way, the `curtis_hack` workaround is not sufficient for correct mappings. Because the controller only understands byte-wise mappings, you have to make two entries, where the second one has a fake subindex of 1. In order not to confuse python-canopen, the second entry is ignored by setting `length=0`, while the first has the complete `length=16` bits specified. Fortunately, Curtis ignores those lengths and uses 8 bit for every entry, making one entry per PDO byte.

```python
o = node.pdo.tx[1]
entry = o.add_variable('Motor_RPM', 0, length=16)
dummy = o.add_variable('Motor_RPM', 1, length=0)
```

Note this only applies to the 123xE series. The 1222 has a very different (and not easily updateable) firmware.

acolomb, 2018-11-22:
See [this gist](https://gist.github.com/acolomb/4c9a613d2f019ad4db5fff34a89bbca7) for my wrapper classes handling the controller-specific quirks. You can derive from them instead of the generic `canopen.Node` class.

af-silva, 2019-10-16:
Hi @Jmeyer1292, can i close this issue??

af-silva, 2019-10-23:
Closing the issue, if needed I can open it again. Thanks

closed 2019-10-23 by af-silva