canopen-python/canopen #129: what is mean of KeyError 5632

Jiongyu, 2018-12-22:
Traceback (most recent call last):
  File "/home/tan/Jupyter-notebook/python/can_test/can_open_test.py", line 49, in <module>
    test()
  File "/home/tan/Jupyter-notebook/python/can_test/can_open_test.py", line 30, in test
    node.rpdo[0x1600][2].phys = 50  #programm current
  File "/usr/local/lib/python3.5/dist-packages/canopen/pdo/base.py", line 36, in __getitem__
    return self.map[key]
  File "/usr/local/lib/python3.5/dist-packages/canopen/pdo/base.py", line 136, in __getitem__
    return self.maps[key]
KeyError: 5632

Jiongyu, 2018-12-22:
I have edit the rpdo1 in the eds file, mapping the 0x2340(programm current) to it .  so I want to tranmit current to motor.

Jiongyu, 2018-12-22:
import  canopen
import time

def print_current(message):
    print("%s received" % message.name)
    for var in message:
        print("%s = %d" % (var.name,var.raw))

def test():
    network = canopen.Network()
    network.connect(channel='can0',bustype = 'socketcan',bitrate=1000000)
    network.sync.start(0.01)
    node = network.add_node(5,'Copley_gripper1.eds')
    node.nmt.state = 'PRE-OPERATIONAL'
    node.rpdo[0x1600][2].phys = 50  #programm current
    node.rpdo[0x1600].start(0.1)
    node.nmt.state = "OPERATIONAL"

    node.tpdo[0x1a00].add_callback(print_current)

    time.sleep(3)
    node.rpdo.stop()

 
    network.sync.stop()

test()

christiansandberg, 2018-12-23:
You should call node.rpdo.read() and node.tpdo.read() to get the maps from the node.

Jiongyu, 2018-12-23:
**My new code :**

import  canopen
import time

def print_current(message):
    print("%s received" % message.name)
    for var in message:
        print("%s = %d" % (var.name,var.raw))

def test():
    network = canopen.Network()
    network.connect(channel='can0',bustype = 'socketcan')
    network.sync.start(0.01)
    node = network.add_node(5,'Copley_gripper1.eds')
    print("network init is over.")

    node.rpdo.read()
    node.rpdo[1].clear()
    node.rpdo[1].add_variable(0x6040)
    node.rpdo[1].add_variable(0x2340)
    node.rpdo[1].enabled = True
    node.rpdo[2].clear()
    node.rpdo[2].add_variable(0x6040)
    node.rpdo[2].add_variable(0x6060)
    node.rpdo[2].enabled = True
    node.rpdo.save()

    node.tpdo.read()
    node.tpdo[1].clear()
    node.tpdo[1].add_variable(0x6041)
    node.tpdo[1].add_variable(0x221c)
    node.tpdo[1].trans_type = 254
    node.tpdo[1].event_timer = 10
    node.tpdo[1].enabled = True
    node.tpdo[2].clear()
    node.tpdo[2].add_variable(0x6041)
    node.tpdo[2].add_variable(0x6061)
    node.tpdo[2].trans_type = 254
    node.tpdo[2].event_timer = 10
    node.tpdo[2].enabled = True
    node.rpdo.save()
    print("rpdo tpdo mapping is over!")

    node.nmt.state = 'PRE-OPERATIONAL'
    node.rpdo[2][0x6060].phys = 4 #set the operation mode as torque mode
    node.rpdo[2].start(0.1)
    node.nmt.state = 'OPERATIONAL'

    flag = 0
    while(flag == 0):
        node.tpdo[2].wait_for_reception()
        mode = node.tpdo[2][0x6061].phys
        if(mode == 4):
            flag = 1
            print("setting the operation mode as torque mode is OK!!")

    node.rpdo[1][0x2340].phys = 50  # mA
    node.rpdo[1].start(0.1)

    for i in rangre(10):
        node.tpdo[1].wait_for_reception()
        current = node.tpdo[1][0x221c].phys
        print(current)
    node.tpdo[1].add_callback(print_speed)

    node.rpdo[1].stop()
    network.sync.stop()
    network.disconnect()
    print("it`s ok")

test()

**but I got a new error**

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/canopen/sdo/client.py", line 68, in read_response
    block=True, timeout=self.RESPONSE_TIMEOUT)
  File "/usr/lib/python3.5/queue.py", line 172, in get
    raise Empty
queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/tan/Jupyter-notebook/python/can_test/canopen_test.py", line 70, in <module>
    test()
  File "/home/tan/Jupyter-notebook/python/can_test/canopen_test.py", line 16, in test
    node.rpdo.read()
  File "/usr/local/lib/python3.5/dist-packages/canopen/pdo/base.py", line 52, in read
    pdo_map.read()
  File "/usr/local/lib/python3.5/dist-packages/canopen/pdo/base.py", line 282, in read
    cob_id = self.com_record[1].raw
  File "/usr/local/lib/python3.5/dist-packages/canopen/variable.py", line 71, in raw
    value = self.od.decode_raw(self.data)
  File "/usr/local/lib/python3.5/dist-packages/canopen/variable.py", line 33, in data
    return self.get_data()
  File "/usr/local/lib/python3.5/dist-packages/canopen/sdo/base.py", line 88, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "/usr/local/lib/python3.5/dist-packages/canopen/sdo/client.py", line 118, in upload
    fp = self.open(index, subindex, buffering=0)
  File "/usr/local/lib/python3.5/dist-packages/canopen/sdo/client.py", line 199, in open
    raw_stream = ReadableStream(self, index, subindex)
  File "/usr/local/lib/python3.5/dist-packages/canopen/sdo/client.py", line 245, in __init__
    response = sdo_client.request_response(request)
  File "/usr/local/lib/python3.5/dist-packages/canopen/sdo/client.py", line 86, in request_response
    return self.read_response()
  File "/usr/local/lib/python3.5/dist-packages/canopen/sdo/client.py", line 70, in read_response
    raise SdoCommunicationError("No SDO response received")
canopen.sdo.exceptions.SdoCommunicationError: No SDO response received

christiansandberg, 2018-12-23:
There is no SDO response from the node for some reason. You may need to inspect the CAN traffic (using candump for example) to see what’s going on.

Jiongyu, 2018-12-24:
**CAN traffic showing above:**
>>$ candump can0
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
  can0  080   [0] 
what is that mean?

Jiongyu, 2018-12-25:
> unexpectedly, One of the signal lines is broken. I fixed it. but I got new error, I have no idea.
> 
> ```
> Traceback (most recent call last):
>   File "/home/tan/Jupyter-notebook/python/can_test/torque_mode.py", line 76, in <module>
>     test()
>   File "/home/tan/Jupyter-notebook/python/can_test/torque_mode.py", line 49, in test
>     node.rpdo[2][0x6071].phys = 4 #set the operation mode as torque mode
>   File "/usr/local/lib/python3.5/dist-packages/canopen/pdo/base.py", line 209, in __getitem__
>     var = self.__getitem_by_index(key)
>   File "/usr/local/lib/python3.5/dist-packages/canopen/pdo/base.py", line 190, in __getitem_by_index
>     value, ', '.join(str(v) for v in valid_values)))
> KeyError: '24689 not found in map. Valid entries are 24640, 24672'
> ```

this question was solved.

closed 2019-10-16 by af-silva