canopen-python/canopen #131: how to do change the mode of operation,such as the position ,velocity,effort etc?

Jiongyu, 2018-12-25:
**### there are  some code to change the mode of operation. I havn`t seen any message regarding to 6040 6060 index in the can traffic. I will appreciate it If you could give a examle to change the mode of opertion.**  

```
    node.nmt.state = 'PRE-OPERATIONAL'
    node.rpdo[4][0x6040].raw = 0x48
    node.rpdo[4].transmit()
    node.rpdo[4][0x6060].raw = 0x4 #set the operation mode as torque mode
    node.rpdo[4].transmit()
    node.rpdo[4].start(0.1)
    node.nmt.state = 'OPERATIONAL'
    print("operation......")

    flag = 0
    while(flag == 0):
        node.tpdo[4].wait_for_reception()
        mode = node.tpdo[4][0x6061].phys
        print(mode)
        if(mode == 4):
            flag = 1
            print("setting the operation mode as torque mode is OK!!")
        else:
            node.rpdo[4][0x6060].phys = 0x4
```

**mode always be zero.**

christiansandberg, 2018-12-25:
Check the examples folder, maybe you will find something useful there?

Jiongyu, 2018-12-26:
I copy the file from the example and run it .  unexcepedly I got a error:
```
node = canopen.Node402(5, 'Copley_T85.eds')
AttributeError: module 'canopen' has no attribute 'Node402'
```

christiansandberg, 2018-12-26:
It seems to be renamed to BaseNode402 now...

Jiongyu, 2018-12-26:
Thank you! 
`AttributeError: 'BaseNode402' object has no attribute 'powerstate_402'`
powerstate_402  change for what?

Jiongyu, 2018-12-27:
**Following the example cia402 , I try to change the mode of operation to torque operation ,but I find the targrt torque cannot sent to Can traffic and the robot still static. I have no idea.   Can you give me some abvice ?  my code present above.**
```
import canopen
import sys
import os
import traceback
import time

def print_torque(message):
    print("%s received" % message.name)
    for var in message:
        print("%s = %d" % (var.name,var.raw))

def test():
    network = canopen.Network()
    network.connect(channel='can0',bustype = 'socketcan')
    network.check()
    #network.connect(bustype='ixxat', channel="can0", bitrate=1000000)
    node = canopen.BaseNode402(5, 'Copley_T85.eds')
    network.add_node(node)
    print("network init is over.")

    # Reset network
    node.nmt.state = 'RESET'
    #node.nmt.state = 'RESET COMMUNICATION'
    node.nmt.wait_for_bootup(15)
    print('node state 1) = {0}'.format(node.nmt.state))

    # Iterate over arrays or records
    error_log = node.sdo[0x1003]
    for error in error_log.values():
        print("Error {0} was found in the log".format(error.raw))

    print("existing node:")
    for node_id in network:
        print (network[node_id])
    print('node state 2) = {0}'.format(node.nmt.state))

    node.sdo[0x1006].raw = 1
    node.sdo[0x100c].raw = 100
    node.sdo[0x100d].raw = 3
    node.sdo[0x1014].raw = 163
    node.sdo[0x1003][0].raw = 0

    # Transmit SYNC every 100 ms
    network.sync.start(0.01)
    node.load_configuration()

    print('node state 3) = {0}'.format(node.nmt.state))
    node.setup_402_state_machine()
    node.state = 'SWITCH ON DISABLED'
    print ('node state 4) = {0}'.format(node.nmt.state))

    # read PDO configuration from ndoe
   
    node.rpdo.read()
    node.rpdo[2]["Control word"].raw = 0x80
    node.rpdo[2].transmit()
    node.rpdo[2]["Control word"].raw = 0x81
    node.rpdo[2].transmit()
    node.rpdo[2]["Modes of operation"].raw = 4 #set the operation mode as torque mode
    node.rpdo[2].transmit()
    node.state = 'READY TO SWITCH ON'
    node.state = 'SWITCHED ON'

    print('Node booted up')

    timeout = time.time() + 15
    node.state = 'READY TO SWITCH ON'
    while node.state != 'READY TO SWITCH ON':
        if time.time() > timeout:
            raise Exception('Timeout when trying to change state')
        time.sleep(0.001)

    timeout = time.time() + 15
    node.state = 'SWITCHED ON'
    while node.state != 'SWITCHED ON':
        if time.time() > timeout:
            raise Exception('Timeout when trying to change state')
        time.sleep(0.001)

    timeout = time.time() + 15
    node.state = 'OPERATION ENABLED'
    while node.state != 'OPERATION ENABLED':
        if time.time() > timeout:
            raise Exception('Timeout when trying to change state')
        time.sleep(0.001)

    print('Node Status {0}'.format(node.state))

    node.nmt.start_node_guarding(0.01)
    while True:
        try:
            network.check()
        except Exception:
            break

        #Read a the mode of operation
        flag = 0
        while (flag == 0):
            mode = node.sdo["Modes of operation display"].raw
            print(mode)
            if (mode == 4):
                flag = 1
                print("setting the operation mode as torque mode is OK!!")
        break

    statusword = node.sdo["Status word"].raw
    print('statusword: {0}'.format(statusword))

    while(1):
        num = input("begin to move? y")
        if(num == 'y'):
            break

    print("operation......")
    node.rpdo[5]["Target Torque"].phys = 20  # mN.m
    node.rpdo[5].transmit()

    while(1):
        node.tpdo[5].wait_for_reception()
        torque = node.tpdo[5]["Torque actual value"].phys
        print("actual torque:{0}".format(torque))

    node.tpdo[5].add_callback(print_torque)
    time.sleep(2)

    print("going to exit!!!")
    if network:
        for node_id in network:
            node = network[node_id]
            node.nmt.stop_node_guarding()
            node.rpdo[5].stop()
            network.sync.stop()
            network.disconnect()
test()
```

Jiongyu, 2018-12-27:
the Can traffic  is  always the following .  the torque always zero.
```
  can0  TX - -  705   [0]  remote request
  can0  RX - -  185   [2]  37 07
  can0  RX - -  385   [6]  37 07 FF FF FF FF
  can0  RX - -  705   [1]  85
  can0  TX - -  080   [0] 
  can0  TX - -  705   [0]  remote request
  can0  RX - -  185   [2]  37 07
  can0  RX - -  385   [6]  37 07 FF FF FF FF
  can0  RX - -  705   [1]  05
  can0  TX - -  080   [0] 
  can0  TX - -  705   [0]  remote request
  can0  RX - -  185   [2]  37 07
  can0  RX - -  385   [6]  37 07 FF FF FF FF
  can0  RX - -  705   [1]  85
```

closed 2019-01-04 by Jiongyu

romainreignier, 2024-10-02:
@Jiongyu I have the same issue, do you remember how did you manage to fix this?