canopen-python/canopen #133: PDO save ro objects

ShatQQ, 2019-01-11:
Hello
I think the pdo save workaround starting at: https://github.com/christiansandberg/canopen/blob/d3f313e2cf4249eb7459b69040376580632a61b0/canopen/pdo/base.py#L349
shall be done for every subindex because maybe the following objects are read only also.

acolomb, 2019-01-11:
Subindex 0 in the array is the length. So we are actually not trying to overwrite an entry, but set the number of entries to zero. That's why the workaround is to fill our mapping array to the expected length.

ShatQQ, 2019-01-14:
Okay I got that one, but what I don't understand also is the next step in `for var in self.map:`.
In this loop you try to overwrite the values in the map_array (`self.map_array[subindex].raw = (var.index << 16 | var.subindex << 8 | var.length)`) with the created dummy values from the filled self.map. And this is also the part i meant. The Object which shall be written in this case can also be read only and the error code 0x06010002 is responded.

acolomb, 2019-01-14:
In that block, we simply don't catch any exceptions.  If a device specifies a length of the mapping array, but then does not allow write access to the specified number of entries, that is a serious protocol violation which is currently not handled.  I don't see a clear way how that **should** be handled, so the code just throws the exception up the caller chain.

Do you have some specific expectation what should happen in case the entries cannot be written?

christiansandberg, 2019-01-14:
I guess that the maps are not possible to be changed during runtime, but you still want to be able to change other properties. It could make sense to continue even if the maps are read-only.

acolomb, 2019-01-14:
But then how does one get notified that the mapping could not be saved? I think an exception is the Right Thing to do in this case. Setting other stuff can always be done manually without using `save()`.

christiansandberg, 2019-01-14:
Hmm, we could continue with the configuration so that it completes but raise the exception in the end so that the application can choose to ignore it if it is an expected error.

acolomb, 2019-01-14:
Sounds reasonable. It's still a workaround for devices that do not actually support mapping at all, so I think the OP could just not call the save method instead. "Writing a read only object" when saving the map is a pretty concise failure description IMHO.

christiansandberg, 2019-01-14:
You make a valid point. I realised that it was more difficult than I originally thought. You can still write the communication parameters manually so it is not a deal breaker. I'll revert my change.

ShatQQ, 2019-01-14:
Hey thanks for the answer. Maybe it would be good to somehow distinguish between saving communication parameters and mapping parameters or both? Or maybe an option to deactivate trying to write a const/ro mapping parameter by checking the od.access_type in the map_array? Right now I chosen the way not to call the save method.

friederschueler, 2024-06-13:
This issue is solved with #273 and subscribing only instead of saving.

commit 3ebd1494 (2019-01-14): Allow PDO maps to be read-only on save()

Fixes #133
closed 2024-06-13 by friederschueler