canopen-python/canopen #136: TPDO subscribed to base COBID in stead of base + nodeID

rubenverhoef, 2019-01-31:
Wen adding an callback to an tpdo it is never called. This because he COBID it is subscribed to is the base in stead of the base+node id. When adding `self.cob_id = self.cob_id + 3` to https://github.com/christiansandberg/canopen/blob/d7d80b158788d85dbec6957c524aee68bccc0195/canopen/pdo/base.py#L283 it is working..

So am i anything missing or is this a bug? Here is my code:
```python
network = canopen.Network()
node = canopen.RemoteNode(3, "Canopen_OD_cfg.eds")
network.add_node(node)

network.connect(bustype='slcan', channel='COM2@57600')
node.tpdo.read()
node.tpdo[1].add_callback(ERROR.Print)

while(True):
    node.nmt.wait_for_heartbeat(timeout = 10)
```

christiansandberg, 2019-02-02:
The COB-ID should be the complete CAN-ID for that PDO. A node can actually have more than 4 PDOs mapped so the ID is not required to include the node ID (3 in this case) but be something completely arbitrary.

What is the manufacturer of the node?

rubenverhoef, 2019-02-04:
Were is the program getting the CAN-ID from?
Is it from the initalization:`node = canopen.RemoteNode(3, "Canopen_OD_cfg.eds")`
or is it from the OD file
or is it requesting from the CAN Node?

The manufacturer of the node is myself with this implementation: https://github.com/CANopenNode/CANopenNode

rubenverhoef, 2019-02-04:
Okay that is strange, what i can see while debugging is:
- The Node is sending it's COB-ID correctly 0x180+node ID (or 0x280+node ID, or whatever)
- When i use the normal code, without my modification the `on_message` function is never called. I also see no reaction from CANOPEN in the debug logging when receving the 0x183 message
- When i Subscribe to te cob-id + 3 it is calling the `on_message` function:
https://github.com/christiansandberg/canopen/blob/d7d80b158788d85dbec6957c524aee68bccc0195/canopen/pdo/base.py#L328
but steps out of it when checking the CANID:
https://github.com/christiansandberg/canopen/blob/d7d80b158788d85dbec6957c524aee68bccc0195/canopen/pdo/base.py#L261

So when i setting the + 3 to this line:
https://github.com/christiansandberg/canopen/blob/d7d80b158788d85dbec6957c524aee68bccc0195/canopen/pdo/base.py#L283
it is working.

Am i something missing? It looks like it does not initializing correctly?

christiansandberg, 2019-02-05:
So when 0x1800:1 is read you get 0x183, but self.cob_id still becomes 0x180?

rubenverhoef, 2019-02-13:
I don't really understand what you mean.
The node is sending PDO 0x1800:1 which results in canid 0x183 but the python canopen does not respond on it. This because it is subscribed to 0x180 in stead of the 0x183. So where is the code where the CANid of the node is used when `node.tpdo[1].add_callback(ERROR.Print)` is called? I Can't find it and when printing the `self.cob_id` in the network.subscribe function it has not included the CANid.

rubenverhoef, 2019-02-13:
Info log gives:
...
```
INFO:canopen.pdo.base:COB-ID is 0x180
INFO:canopen.pdo.base:PDO is enabled
INFO:canopen.pdo.base:RTR is allowed
INFO:canopen.nmt:Received heartbeat can-id 1795, state is 5
INFO:canopen.pdo.base:Transmission type is 255
INFO:canopen.pdo.base:Inhibit time is set to 100 ms
INFO:canopen.pdo.base:Event timer is set to 0 ms
INFO:canopen.pdo.base:SYNC start value is set to 0 ms
INFO:canopen.pdo.base:Adding Manufacturer status register (0x1002:0, 32 bits) to PDO map

INFO:canopen.pdo.base:COB-ID is 0x280
INFO:canopen.pdo.base:PDO is enabled
INFO:canopen.pdo.base:RTR is allowed
INFO:canopen.pdo.base:Transmission type is 254
INFO:canopen.pdo.base:Inhibit time is set to 0 ms
INFO:canopen.pdo.base:Event timer is set to 0 ms
INFO:canopen.pdo.base:SYNC start value is set to 0 ms

INFO:canopen.pdo.base:COB-ID is 0x380
INFO:canopen.pdo.base:PDO is enabled
INFO:canopen.pdo.base:RTR is allowed
INFO:canopen.pdo.base:Transmission type is 254
INFO:canopen.pdo.base:Inhibit time is set to 0 ms
INFO:canopen.pdo.base:Event timer is set to 0 ms
INFO:canopen.pdo.base:SYNC start value is set to 0 ms

INFO:canopen.pdo.base:COB-ID is 0x480
INFO:canopen.pdo.base:PDO is enabled
INFO:canopen.pdo.base:RTR is allowed
INFO:canopen.nmt:Received heartbeat can-id 1795, state is 5
INFO:canopen.pdo.base:Transmission type is 254
INFO:canopen.pdo.base:Inhibit time is set to 0 ms
INFO:canopen.pdo.base:Event timer is set to 0 ms
INFO:canopen.pdo.base:SYNC start value is set to 0 ms
```
....

rubenverhoef, 2019-02-13:
Problem found:
It is getting the NodeID from the `1800subx DefaultValue` from EDS file which is `$NODEID+0x180`, when i hard code it to `0x183` it works.
Is the `$NODEID` value in the EDS supported?

rubenverhoef, 2019-02-13:
I see why it is not working, the python canopen code is getting the COB-ID from 1800sub1 of the node. It requests the 1800sub1 and use it to subscribe to. But my device sends the standard 0x180 without the nodeid part. Should my node contain at SDO 1800sub1 0x180 + nodeID or is this a bug of Python CANopen?

acolomb, 2019-02-13:
The node must definitely return the **final** COB ID in parameter `0x1800:01`.  If using the "predefined connection set", that is the base value `0x180` + node ID.

If your node sends its PDO with a COB ID different from what the parameter says, it is very broken.

rubenverhoef, 2019-02-14:
Okay, that is clear. So this is not an issue of python CANopen so i will close. I will look into my node, it is my own code using: https://github.com/CANopenNode/CANopenNode. But now i know what's going wrong and where to look. Thanks all

closed 2019-02-14 by rubenverhoef