canopen-python/canopen #139: eds file

Tanrikulu, 2019-02-03:
Hello,

the eds-file I used seems to be defined in a different way.
When I add "vendor_id = node.sdo[0x1018][1].raw " into my code, I get "0x1018 was not found in Object Dictionary".

This is the structure of my eds-file:
```
[1018]
ParameterName=Identity object
ObjectType=8
SubNumber=4

[1018sub0]
ParameterName=Number of entries
ObjectType=7
DataType=0x5
AccessType=ro
DefaultValue=3
PDOMapping=0

[1018sub1]
ParameterName=Vendor Id
ObjectType=7
DataType=0x7
AccessType=ro
DefaultValue=0x00000139
PDOMapping=0

[1018sub2]
ParameterName=Product code
ObjectType=7
DataType=0x7
AccessType=ro
DefaultValue=0
PDOMapping=0

[1018sub3]
ParameterName=Revision number
ObjectType=7
DataType=0x7
AccessType=ro
DefaultValue=0
PDOMapping=0
```
How can I adjust my eds-file to the canopen-code?

christiansandberg, 2019-02-04:
0x1018 should be a Record (object type 9), not an Array (8).

```
[1018]
ParameterName=Identity object
ObjectType=9
SubNumber=4
```

Tanrikulu, 2019-02-06:
You are right. Thanks

closed 2019-02-06 by christiansandberg