canopen-python/canopen #140: control two canopen-motors

Tanrikulu, 2019-02-06:
Hello,

I would like to control two canopen-motors indepedently. I am using one sensor per motor.

My question is:
Should I use threads in my code to control the motors independently? When I would not use any threads, the program would only process the sensor data of one motor. Right?
Has anyone an idea how I can drive the motors simultameously?

christiansandberg, 2019-02-06:
Are you controlling them using SDO or PDO?

Tanrikulu, 2019-02-06:
Thank you for your feedback.
I will start to develop the control software tomorrow. I am not sure.
But what difference would that make regarding this issue, if I use SDO or PDO?

Tanrikulu, 2019-02-11:
What would you recommend? Controlling the motors with SDO or PDO?

af-silva, 2019-02-14:
The recommended way it's to control by PDO. If you read a bit about the canopen protocol you will see that SDO acts more like a the TCP protocol (not a direct analogy here) and the PDO as the UDP protocol. That being said, the SDO transmit a request to the controller and the controller respond's unlike the PDO that are broadcast from one node to another. PDO's can be configured in each remote node to broadcast it's status and data about the controller/motor (e.g., velocity, number of revolutions, consumption, ...) to the master what reduces the number of messages that are present at any one time in the BUS, avoiding collisions and other messing stuff. Normally each motor needs to publish this information at a very high rate so this is important. The SDO's on the other hand are normally use at start-up to initialize the controller/motor since one needs to receive confirmation that everything went ok.

You don't need to worry about threads here, the code access the can bus and reads and writes without any problem. If you want you may read more about [here](https://python-can.readthedocs.io/en/stable/bus.html#thread-safe-bus), and also it depends on the CAN-Interface you are using, for instance the Kvaser only allows one handle per thread  *"Two separate handles are needed if receiving and sending messages are done in different threads"* - read [here](https://python-can.readthedocs.io/en/stable/interfaces.html).

I'm using SocketCan and only have on thread running the code and everything is fine, no problem so far. I'm controlling to three motors and the response is prime ;)
@christiansandberg correct me if I'm wrong

Tanrikulu, 2019-02-15:
Thank you very much for your support.

christiansandberg, 2019-02-15:
Well said!

af-silva, 2019-02-18:
@Tanrikulu No problem, I hope I helped :) if you don't have mor questions regarding this and if your problem has been resolved please  close the ticket.

closed 2019-02-19 by Tanrikulu