canopen-python/canopen #141: Error in node.pdo.save()

zcchen, 2019-02-12:
Hello.

I got the error messages as below:
```
Traceback (most recent call last):
  File "testing.py", line 55, in <module>
    motor_node.tpdo.save()
  File "/usr/local/lib/python3.7/site-packages/canopen/pdo/base.py", line 57, in save
    pdo_map.save()
  File "/usr/local/lib/python3.7/site-packages/canopen/pdo/base.py", line 334, in save
    self.com_record[1].raw = self.cob_id | PDO_NOT_VALID
TypeError: unsupported operand type(s) for |: 'NoneType' and 'int'
```
with the codes like below:
```
      node.tpdo[1].clear()
      node.tpdo[1].add_variable('8 bit digital input block area', '1. digital input block')
      node.tpdo[1].add_variable('8 bit digital input block area', '2. digital input block')
      node.tpdo[1].enabled = True
      node.tpdo.save()
```
I have checked some variables,
`node.tpdo[1].cob_id` is `None`,
and `node.tpdo[1].predefined_cob_id` is `0x181` (my node id is `1`).

Did I misunderstand something or should something be improved?
Thanks very much.

acolomb, 2019-02-12:
You should explicitly set a COB ID for the object, unless you are calling its `read()` method.

Just use `node.tpdo[1].cob_id = node.tpdo[1].predefined_cob_id` to use the default.

zcchen, 2019-02-13:
I did try as above, and the `cob_id` had been modified to the default value, but this error did not disappear.

zcchen, 2019-02-13:
I have located why this error happened.
The function `PdoBase.save()` in the file `canopen/pdo/base.py` saves the settings of all PDOs (PDO1  - PDO4) including the PDOs  which are not activated by user.
Not all PDOs are used in engineering, thus  I suggest change the `PdoBase.save()` method, ignoring the cleared PDO.

coldicecream, 2019-08-30:
@zcchen Did you have any resolution to this?

af-silva, 2019-10-16:
Hi @zcchen,
You need to read first 
``` python
node.tpdo.read()
node.rpdo.read()
```

Then you can access the dictionaries, and set the properties you want. Do you have an eds file??

af-silva, 2019-10-23:
Closing the issue, if needed I can open it again. Thanks

closed 2019-10-23 by af-silva

zcchen, 2019-11-08:
Sorry for my late reply.

1, I found this issue, due to the situation that there is no eds from
devices' suppliers. And it is a common issue when we use some new devices
which is still being developed.

2, Currently, I have reimplemented another pdo methods with sdo only, which
is not related to the eds file. I supposed that the developers know what he
is doing. It is just a workaround solution.

3, In my opinion, I don't think we need to do everything based on eds file.
If no eds file, that would be in chaos.

af-silva, 2019-11-09:
Thanks for the reply @zcchen,

> CANopen electronic data sheet specification for CANopen (EDS)
> 
> The usage of devices in a communication network requires configuration of the device parameters and communication facilities. CANopen defines a standardised way to access these parameters via the object dictionary.
>
> For handling of the complexity of CANopen systems Software Tools are required. This reduces the complexity of the planning, configuration and analysis process and significantly increases the security o the system.
>
> For this purpose Software Tools need an electronic description of the CANopen devices. To allow the usage of manufacturer independent Tools, this document defines a standardised file format – called Electronic Data Sheet EDS.
>
> Furthermore some derived file formats are specified. The DCF describes a concrete incarnation of a device configuration. The MDS describes modules of devices with a modular structure.
>
>[Source ](http://overpof.free.fr/schneider/CAN%20&%20CANopen/CANopen/%A9CiA%20CANCANopen%20CD%20V5.1/html/spec3.htm#eds)

Reading this **_CANopen defines a standardised way to access these parameters via the object dictionary_** **_... defines a standardised file format – called Electronic Data Sheet EDS._**, so in my humble opinion I think EDS files (or DCF) are good for standardization and since this API should be agnostic of any custom implementations a think it's a god design choice. Like you said, _"I supposed that the developers know what he is doing"_ I think you are right, and they have access to the code and can tinker with it, and adapt the implementation for their needs.

Also, the EDS file works has a data store for the Object Dictionary, please refer to [this EDS](https://github.com/CANopenNode/CANopenNode/blob/master/example/IO.eds) file from [CANopenNode](https://github.com/CANopenNode/CANopenNode) so if you use only a sub part of a specific Object Dictionary you can create a file similar to this one with only the objects you need. And it's a plus if you need to change a specific value i don't need to go inside the code change it...
@christiansandberg Can also give his opinion.

@zcchen Do you need me to reopen this discussion or can I assume this topic as closed?

christiansandberg, 2019-11-09:
I have actually been thinking about modifying the PDO code to not rely on the object dictionary being correctly set up. That would unfortunately involve a lot of packing and unpacking of raw data and since I believe most devices have an EDS file I have not prioritized that.

zcchen, 2020-01-05:
Sorry for my late reply. I think we can assume this topic as closed.

Most devices have EDS files usually for the product version. But as a R&D engineer, I have to touch many under-developing devices which do NOT have any EDS files.
Meanwhile, some device suppliers in China provide incorrect EDS files but correct manuals for CANopen dictionaries.
Thus, we have to avoid any EDS files in our production codes.

That was why I would like to have this feature for some spike testing.
If we have this feature in future, that would be great.
If no, there are also OK since I have implemented a workaround solution.
If I have time, I can try to contribute my codes to this repo.

Thanks very much for everyone.

af-silva, 2020-01-08:
Ok, you are welcome @zcchen to contribute.
Keep hacking! xD