canopen-python/canopen #142: E1136:Value 'nodeRight.sdo[14852]' is unsubscriptable

Tanrikulu, 2019-02-15:
Hello all,

I would like to read the measured velocity of my motor by using this command:
`nodeLeft.sdo[0x3A04][1].phys`

But I get an error-message:
`[pylint] E1136:Value 'nodeLeft.sdo[14852]' is unsubscriptable`

My eds-file indludes this object:
```
[3A04sub1]
ParameterName=Measured velocity in [rpm]
ObjectType=7
DataType=0x4
AccessType=ro
DefaultValue=
PDOMapping=1

```
And when I put the whole paramter name into the brackets, I get an error message like "Measured velocity in [rpm] is not found in objectdictionary"

I do not undertand that. Has anyone an idea?

christiansandberg, 2019-02-15:
What does [3A04] look like in the EDS file?

Tanrikulu, 2019-02-15:
Hi,

this is from the eds file.
```
[3A04]
ParameterName=Measured velocity in increments
ObjectType=9
SubNumber=2

[3A04sub0]
ParameterName=Measured velocity in increments
ObjectType=7
DataType=0x4
AccessType=ro
DefaultValue=
PDOMapping=1

[3A04sub1]
ParameterName=Measured velocity in [rpm]
ObjectType=7
DataType=0x4
AccessType=ro
DefaultValue=
PDOMapping=1

```

Tanrikulu, 2019-03-02:
This is the eds file I use. 
[EDS.txt](https://github.com/christiansandberg/canopen/files/2921983/EDS.txt)

In case I access an object through the subindex, I get the error-message like `[pylint] E1136:Value 'nodeLeft.sdo[14852]' is unsubscriptable`. I also tried to access the object via the parameter name, but the same error occurs.

christiansandberg, 2019-03-03:
Since it contains the invalid UTF-8 character I had to strip out some stuff, but 3A04 was still left and I had no problems with it. How did you get pass the encoding problem?

Tanrikulu, 2019-03-04:
I have only removed the last three lines in the eds file. 
This lines:
```
[Comments]
Lines=1
Line1=EDS File for the miControl(R) DSA-Devices
```

In code I have added:
```
eds1 = open(file='/home/pi/.local/lib/python3.5/site-packages/canopen/mcDSA-Bxx_test1.eds', mode='r', encoding='utf-8')
eds2 = open(file='/home/pi/.local/lib/python3.5/site-packages/canopen/mcDSA-Bxx_test2.eds', mode='r', encoding='utf-8')

nodeLeft = canopen.RemoteNode(64, eds1)
network.add_node(nodeLeft)
nodeRight = canopen.RemoteNode(80, eds2)
network.add_node(nodeRight)
```

Tanrikulu, 2019-03-04:
I get this error only for objects that require a subindex. For objects that do not require a subindex, it works fine. For example I do not need a subindex to set the desired velocity.
```
[3300]
ParameterName=Velocity - desired value
ObjectType=7
DataType=0x4
LowLimit=-2147483648
HighLimit=2147483647
AccessType=rw
DefaultValue=0
PDOMapping=1
```

christiansandberg, 2019-03-05:
And it's not only a Lint warning? You also get an exception when you execute the code?

af-silva, 2019-10-16:
@Tanrikulu did you find a solution?

af-silva, 2019-10-23:
Closing the issue, if needed I can open it again. Thanks

closed 2019-10-23 by af-silva