canopen-python/canopen #145: PDO reception should also update the values in the dictionary

grw1983, 2019-03-11:
I'm implementing a simple node using "canopen.LocalNode()". The node has some PDOs defined. I noticed, that while continuously reading the object values using "node.sdo[0x2002].raw", setting this object via the CANOpen master using a PDO, it doesn't change the value of the object. When writing 0x2002 via SDO write, this works as expected.

Also connecting a "PDO" callback to the object works as expected.

Is the idea here, to use the "PDO" callback to set the value in the object dictionary?

christiansandberg, 2019-03-13:
No you are right, PDO slave support is not fully implemented. Currently it is just made to work like the master.

grw1983, 2019-03-14:
Can you coarsely describe how you would like to see this implemented?

I'm thinking of just adding this functionality to the Map::on_message() function, after all the callbacks returned okay.