canopen-python/canopen #148: Support 29 bits COB-ID

HyunYoung-Kim, 2019-03-14:
Hey Christian
Thank for your effort for developing CANopen stack for Python.

I have a device which uses four SDO server channels.
the first one is the standard SDO server: 0x600 /0x580 + node-id, defined in object 0x1200
the second one is the standard SDO server: 0x640 /0x5C0 + node-id, defined in object 0x1201
the third one is the standard SDO server: 0x1C000000 /0x18000000 + node-id, define in 0x1202
the last one is the standard SDO server: 0x1D000000 /0x19000000 + node-id, defined in 0x1203

But you are using just the standard SDO channel.
' self.sdo = SdoClient(0x600 + self.id, 0x580 + self.id, self.object_dictionary)'

I tried to change the fix-offset in source-code. I think, your stack doesn't support 29bits.
Are you going to change it?
kind regard Hyun-Young

christiansandberg, 2019-03-14:
You are right that it does not support 29-bit identifiers yet. I was not sure if anyone actually used that.

I can have a look and see if it is doable without too much work involved.

HyunYoung-Kim, 2019-03-14:
Hey Christian
Thank a lot.
Hyun-Young

 
 
 

repository","main_image_url":"https://github.githubassets.com/images/email/message_cards/header.png","avatar_image_url":"https://github.githubassets.com/images/email/message_cards/avatar.png","action":{"name":"Open 
in 
GitHub","url":"https://github.com/christiansandberg/canopen"}},"updates":{"snippets":[{"icon":"PERSON","message":"@christiansandberg 
in #148: You are right that it does not support 29-bit identifiers yet. I 
was not sure if anyone actually used that.\r\n\r\nI can have a look and see 
if it is doable without too much work involved."}],"action":{"name":"View 
Issue","url":"https://github.com/christiansandberg/canopen/issues/148#issuecomment-473004131"}}} 
"https://github.com/christiansandberg/canopen/issues/148#issuecomment-473004131", 
"url": 
"https://github.com/christiansandberg/canopen/issues/148#issuecomment-473004131", 
"name": "View Issue" }, "description": "View this Issue on GitHub", 
"publisher": { "@type": "Organization", "name": "GitHub", "url": 
"https://github.com" } } ]

christiansandberg, 2019-03-20:
I implemented a simple solution for now where IDs > 0x7FF will be sent as extended ID.

closed 2019-03-20 by christiansandberg

HyunYoung-Kim, 2019-03-20:
Thank a lot!