canopen-python/canopen #152: General question

Wez50, 2019-04-01:
hello, I have had everything working for over a year without issue, but i have been forced to update my image with the beaglebone black, and a lot has changed. i can see my servo drive hearbeat with Candump but i am thinking i have some sort of a disconnect with my python version ? i have "python 2.7.13" as the version the beaglebone uses if i use 'python -V'...... but i can only get the scripts to run by typing python3,

these are examples of the types of errors i am getting
any ideas for what i need to investigate would be fantastic, as i fear i am simply starting to mess around without understanding the problem.

`debian@beaglebone:~/my_python$ python3 HelloCan.py
Traceback (most recent call last):
  File "HelloCan.py", line 37, in <module>
    node.pdo.rx[3]['Controlword', 'Target position'].phys = 1000
  File "/home/debian/.local/lib/python3.5/site-packages/canopen/pdo/base.py", line 212, in __getitem__
    var = self.__getitem_by_index(int(key, 16))
TypeError: int() can't convert non-string with explicit base

`
`
debian@beaglebone:~/my_python$ python3 HelloCan.py
Traceback (most recent call last):
  File "HelloCan.py", line 33, in <module>
    node.pdo.export('database.dbc')
  File "/home/debian/.local/lib/python3.5/site-packages/canopen/pdo/base.py", line 68, in export
    from canmatrix import canmatrix
ImportError: No module named 'canmatrix'

`debian@beaglebone:~/my_python$ python3.5 MoveCan.py
Traceback (most recent call last):
  File "MoveCan.py", line 3, in <module>
    from canopen.profiles.p402 import Node402
ImportError: cannot import name 'Node402'

debian@beaglebone:~/my_python$ python3.5 MoveCan.py
Traceback (most recent call last):
  File "MoveCan.py", line 4, in <module>
    node = canopen.Node402(32, 'object_dictionary.eds')
AttributeError: module 'canopen' has no attribute 'Node402'

debian@beaglebone:~/my_python$ python3.5 MoveCan.py
Traceback (most recent call last):
  File "MoveCan.py", line 6, in <module>
    network.add_node(node)
NameError: name 'node' is not defined

debian@beaglebone:~/my_python$ python3.5 MoveCan.py
Traceback (most recent call last):
  File "MoveCan.py", line 9, in <module>
    node.setup_402_state_machine()
NameError: name 'node' is not defined
`
`

Wez50, 2019-04-02:
Ok so my problem is my old code isnt inline with all the changes that have happen (sorry)........ i assumed that everything had remained the same with canopen, obviously alot has changed since i last installed it. Feb 2018

I belive i have comms as i can see my drive LED flash and then become steady 'on' + i can see the heatbeat in candump.

i am working through an example code but i just cant get passed a problem with how i am using the state machine this is the runtime error i get :- 

`root@beaglebone:/home/debian/my_python# python3 wesscan.py
node state 1) = PRE-OPERATIONAL
node state 2) = PRE-OPERATIONAL
node state 3) = PRE-OPERATIONAL
emcl
668
Traceback (most recent call last):
  File "wesscan.py", line 26, in <module>
    node.setup_402_state_machine()
  File "/home/debian/my_python/canopen/canopen/profiles/p402.py", line 232, in setup_402_state_machine
    self.state = 'SWITCH ON DISABLED'
  File "/home/debian/my_python/canopen/canopen/profiles/p402.py", line 456, in state
    raise RuntimeError('Timeout when trying to change state')
RuntimeError: Timeout when trying to change state

`

**this is the code i am running basic example copy to try and get things moving:**

`import time
import canopen
import sys
import os
import traceback
network = canopen.Network()
network.connect(channel='can1', bustype='socketcan')
network.check()
node = canopen.BaseNode402(32, '/home/debian/my_python/object_dictionary.eds')
network.add_node(node)

node.nmt.state = 'RESET COMMUNICATION'
node.nmt.wait_for_bootup(15)
print('node state 1) = {0}'.format(node.nmt.state))
error_log = node.sdo[0x1003]

print ('node state 2) = {0}'.format(node.nmt.state))
network.sync.start(0.1)

node.load_configuration()
print('node state 3) = {0}'.format(node.nmt.state))
device_name = node.sdo[0x1008].raw
vendor_id = node.sdo[0x1018][1].raw
print(device_name)
print(vendor_id)
node.setup_402_state_machine()
node.sdo[0x6040].raw = 0x06
node.sdo[0x6041].raw
#node.state = 'SWITCH ON DISABLED'

timeout = time.time() + 15
node.state = 'READY TO SWITCH ON'
while node.state != 'READY TO SWITCH ON':
        if time.time() > timeout:
                raise Exception('Timeout when trying to change state')
time.sleep(0.001)

timeout = time.time() + 15
node.state = 'SWITCHED ON'
while node.state != 'SWITCHED ON':
        if time.time() > timeout:
                raise Exception('Timeout when trying to change state')
time.sleep(0.001)

timeout = time.time() + 15
node.state = 'OPERATION ENABLED'
while node.state != 'OPERATION ENABLED':
 if time.time() > timeout:
                raise Exception('Timeout when trying to change state')
time.sleep(0.001)

`

af-silva, 2019-04-02:
Hi, 

First things first, can you please provide the content of the EDS file??
The error you get seems to me that the response from the driver never reaches the Master Node (`RuntimeError: Timeout when trying to change state`). 

Looking at your code you are using the following configuration:
`node = canopen.BaseNode402(32, '/home/debian/my_python/object_dictionary.eds')`
The node ID for the motor you are trying to reach is 32 (dec) = 20 (hex) and the file is at `/home/debian/my_python`. The node ID is correct??

The `setup_402_state_machine()` setups the state machine for the node, it needs the Controlword and Statusword to be configured correctly, and then it needs the motor no handle the DS 402 state machine

[image]
( credits to [Nanotec](https://en.nanotec.com/products/manual/N5_ECAT_EN/general%252Fds402_power_state_machine.html/) )

Make sure the motor is well configured and use the candump to see if the motor is responding to the commands send by the ds402 node.

Hope this helps...

Wez50, 2019-04-02:
Thank you for taking the time to advise and respond, i really appreciate the support.
generally once i get a start i am able to 'get going' / 'experiment' to learn 

for the .eds file i have :- 
[2000sub1]
ParameterName=Node ID
ObjectType=0x7
DataType=0x0005
AccessType=rw
DefaultValue=**0x20**
PDOMapping=0x0
LowLimit=0x1
HighLimit=0x7F

**within the drive i have**

TPDO1  
Index = 0x6041
Subindex = 0x00
Name = Statusword
COB-ID = 1A0
transmission type = Event driven
event time (ms) = 0
inhibit time (ms) = 100  

RPDO1
Index = 0x6040
subindex = 0x00
Name = Controlword
COB-ID = 220
transmission type = async

Candump (well the last section )
 can1  000   [2]  01 20
  **can1  1A0   [2]  21 02**
  can1  2A0   [3]  21 02 06
  can1  3A0   [6]  21 02 00 00 00 00
  can1  4A0   [6]  21 02 00 00 00 00
  **can1  1A0   [2]  21 02**
  can1  2A0   [3]  21 02 06
  can1  3A0   [6]  21 02 00 00 00 00
  can1  4A0   [6]  21 02 00 00 00 00
  **can1  220   [2]  00 00**
  can1  080   [0]
  can1  080   [0]
  can1  080   [0]
  can1  720   [1]  05
  can1  080   [0]
  **can1  220   [2]  00 00**

this is the current code i am using now, to almost copy the example 

`import time
import canopen
import sys
import os
import traceback
network = canopen.Network()
network.connect(channel='can1', bustype='socketcan')
network.check()
node = canopen.BaseNode402(32, '/home/debian/my_python/object_dictionary.eds')
network.add_node(node)

node.nmt.state = 'RESET COMMUNICATION'
node.nmt.wait_for_bootup(15)
print('node state 1) = {0}'.format(node.nmt.state))
error_log = node.sdo[0x1003]

print ('node state 2) = {0}'.format(node.nmt.state))
network.sync.start(0.1)
node.load_configuration()
print('node state 3) = {0}'.format(node.nmt.state))

node.setup_402_state_machine()
device_name = node.sdo[0x1008].raw
vendor_id = node.sdo[0x1018][1].raw
print(device_name)
print(vendor_id)
node.state = 'SWITCH ON DISABLED'

`

Wez50, 2019-04-03:
i have managed to get the code to run partly.
My current block on progress relates to the following 2 errors :- 
1) for some reason i can not write to the database.dbc.
2) if i delete the database export line, the code will run to the end without error. but i get no motor movement i have had great difficulty trying to find out how to use the sudo code in the documents to use rpdo or sdo so i am very unsure of the best way forward.

THE DATABASE ISSUE:-
`Traceback (most recent call last):
  File "wesscan.py", line 73, in <module>
    node.rpdo.export('database.dbc')
  File "/home/debian/my_python/canopen/canopen/pdo/base.py", line 80, in export
    extended=0)
TypeError: __init__() got an unexpected keyword argument 'Id'`

if i delete the line exporting to database.dbc the code runs but the motor doesn't move.... but i am finding it very difficult to get the syntax correct from the documentation.
I have attempted many different combinations but i have included two examples below for what i believe should move the motor. (obviously i am way off the correct method)

MOTOR MOVE CODE ISSUE:-
`time.sleep(0.8)
node.sdo[0x6040].raw = 0x0F
node.rpdo[4]['Target position'].phys = 10000
node.rpdo[4]['Target velocity'].phys = 50000
node.sdo[0x6040].raw = 0x1F
node.rpdo[4].start(0.1)

time.sleep(0.8)
node.sdo['Controlword'].raw = 0x0F
node.sdo['Target position'].raw = 300000
node.sdo['Target velocity'].raw = 5000
`
any guidance will be appreciated
thanks
wess

af-silva, 2019-04-08:
Hi, that is strange... You don't need to export, the export is to export da configurations to a ´.dbc´ file.
What you need is to first configure the nodes you are trying to control like

```python
# Add some nodes with corresponding Object Dictionaries
node1 = canopen.BaseNode402(35, '<Path to eds file>')
node2 = canopen.BaseNode402(35, '<Path to eds file>')
network.add_node(node1)
network.add_node(node2)

...
for id in self.network.nodes:
  node = self.network[id]
  # Load the configuration read from the EDS file to all the nodes
  node.load_configuration()
  # Start the internal DS402 State Machine
  node.setup_402_state_machine()
  # Start the node guarding mechanism
  node.nmt.start_node_guarding(0.01)
  node.state = 'OPERATION ENABLED'
```

If all the configuration nodes are present in the eds file you don't need to do the configuration by hand like in the code example (e.g., `node.sdo[0x6040].raw = 0x0F`), make sure to call the ` node.load_configuration()`, to make sure the configuration for each node has been loaded correctly you can call the `node.rpdo.read()` and `node.tpdo.read()`. 

One important thing, the **Controlword** and **Statusword** must be well configure. The code used in the example ds402 was used by me to test the correct control of my motor. 
In my configuration I have to configured the heartbeat (the NMT state Guard) the ensures the nodes are running, if the master can't reach the nodes, they will shutdown. 

EDS example, some of the values
```
[6040]
ParameterName=Controlword
ObjectType=0x7
DataType=0x0006
AccessType=rww
DefaultValue=0x0
PDOMapping=0x1

...

[1600]
SubNumber=0x9
ParameterName=RPDO 1 mapping parameter
ObjectType=0x9

[1600sub0]
ParameterName=number of mapped objects
ObjectType=0x7
DataType=0x0005
AccessType=rw
PDOMapping=0x0
ParameterValue=0x2

[1600sub1]
ParameterName=Target Velocity
ObjectType=0x7
DataType=0x0007
AccessType=rw
DefaultValue=0x60ff0020
PDOMapping=0x0
ParameterValue=0x60ff0020

[1600sub2]
ParameterName=Control Word
ObjectType=0x7
DataType=0x0007
AccessType=rw
DefaultValue=0x60400010
PDOMapping=0x0
ParameterValue=0x60400010

...

[1A00]
SubNumber=0x9
ParameterName=TPDO 1 mapping parameter
ObjectType=0x9

[1A00sub0]
ParameterName=number of mapped objects
ObjectType=0x7
DataType=0x0005
AccessType=rw
DefaultValue=0x2
PDOMapping=0x0
ParameterValue=0x2

[1A00sub1]
ParameterName=Velocity Actual Value
ObjectType=0x7
DataType=0x0007
AccessType=rw
DefaultValue=0x606C0020
PDOMapping=0x0
ParameterValue=0x606C0020

[1A00sub2]
ParameterName=Status Word
ObjectType=0x7
DataType=0x0007
AccessType=rw
DefaultValue=0x60410010
PDOMapping=0x0
ParameterValue=0x60410010

```
The object 6040 is the ControlWord and the object 6041 is the StatusWord. They are mapped to the RDPO1 and TPDO1 like you can see in the configuration above. My RPDO1 maps the **Controlword** and the **Target velocity**, and my TPDO1 maps the **Statusworkd** and the **Velocity Actual Value**. This objects are used to issue commands to the drive, also the drive is configured to take action, this is, try to achieve the target velocity at each heart beat sent from the master node (0x80).
Also, you need to send using the **Controlword**  the instruction for the drive to move right away, like 

```python
SW_TARGET_REACHED = 0x400
SW_SETPOINT_ACKNOWLEDGE = 0x1000
SW_FOLLOWING_ERROR = 0x2000

CW_NEW_SETPOINT = 0X10
CW_NEW_SETPOINT_CHANGE_IMMEDIATELY = 0X30
CW_CHANGE_IMMEDIATELY = 0x20
CW_HALT = 0x100

...

def target_position(self, value):
	"""Target Position Setter used to set the target position of the node
	:param value: Position in counts (device internal units)
	"""
	try:
		self.rpdo_pointers[0x607A].raw = value
		self.rpdo_pointers[0x607A].pdo_parent.transmit()
		self.controlword = State402.CW_OPERATION_ENABLED | self.CW_NEW_SETPOINT_CHANGE_IMMEDIATELY
		timeout = time.time() + 0.5
		while (self.statusword & self.SW_SETPOINT_ACKNOWLEDGE) != self.SW_SETPOINT_ACKNOWLEDGE:
			time.sleep(0.0001)
			if time.time() > timeout:
				raise RuntimeError('Unable to set new position')
		self.controlword = State402.CW_OPERATION_ENABLED | self.CW_CHANGE_IMMEDIATELY
	except KeyError as ex:
		raise ValueError('The object 0x607A (Target Position) is not configured in any PDO, please check the device configuration! Error[{0}]'.format(ex.__str__()))

```
In the example above I post some of my code to set the target position of the drive using the State402 class and BaseNode402 class ... I can't however post more code ... sorry.

Also make sure your BUS configuration is OK like: `{'bustype': 'socketcan', 'channel': 'slcan0', 'bitrate': 1000000 }`

To test, please read using the candump the messages that are being sent in the network bus. The candump is a tool provided by the SocketCAN utils [here](https://elinux.org/Can-utils).

Sorry if I don't explained my self well, I don''t have much time now, but please try to see if you are reaching the nodes through the candump and check your configuration. Also don't look only at the example, it was meant to only show how to start working with the code and access the controllers. The State402 class has a lot of code that helps in the communication with the controllers but you need to read it and understand how they work.

Wez50, 2019-04-11:
Sorry for the delay in response but i wanted to fully understand all the fantastic information you have provided. I was able to get it working through brute force and ignorance some days ago, however you have highlighted so many aspects i obviously need much more time to work through and test to gain understanding and apply what you have passed on to me..... it makes a lot of sense now i can see an example.

while i have this line open could i just get your advice about status words when you have time, last year i was homing my motor i run into some difficulty with looking at individual bits for "homing active' ' homing achived' etc and ended up with some messy code to bitshift and use a 'AND' operator to check a statusword bit in a loop...... probably not the best way to do it.
 am i correct in thinking (by reading the code above)  all the changes that have been made in this latest revision enable me to bit mask and assign a variable for these type of tasks like homing and looking for changes in individual bits? for example you used :

SW_SETPOINT_ACKNOWLEDGE = 0x1000
while (self.statusword & self.SW_SETPOINT_ACKNOWLEDGE) != self.SW_SETPOINT_ACKNOWLEDGE:

my project is not for any real 'job' or reason, i am simply very interested in Canopen and enjoy learning something new. i have years with controls but only just starting again with python / canopen from my attempts last year, unfortunately i had to stop due to some family reasons. 

Thank you for your time and patience. 
Wess

af-silva, 2019-04-12:
Hi,

No problem, I'm glad to help. I read my answer again and sorry for messing the explanation with different types of control, but the principal is the same. In the configuration example I was configuring the RPDO and TPDO for velocity control, and in the code below I showed how to control in position mode xD.

> am i correct in thinking (by reading the code above) all the changes that have been made in this latest revision enable me to bit mask and assign a variable for these type of tasks like homing and looking for changes in individual bits?

Yes, the DS402 profile is mostly handle by the class **BaseNode402** and you just need to extend this and implement the features you need, like my code example above. No need to bitshift, you can just use the masks like: 

```python
while (self.statusword & self.SW_TARGET_REACHED) != self.SW_TARGET_REACHED: 

...

if ((self.statusword & self.SW_TARGET_REACHED) == self.SW_TARGET_REACHED and
                (math.fabs(self.position_window) - 100 < math.fabs(target_position) < math.fabs(self.position_window) + 100)):

...

elif (self.statusword & self.SW_TARGET_REACHED) == self.SW_TARGET_REACHED:

...

```

The masks are configured per type of control (e.g., Position, Velocity, Torque) and you need to define this yourself. The API implement here is generic, implementing only the state machine and core functions and enabling easy configuration/handle of the nodes.

Also, when trying to **Home** the drive, you need to check the configuration that is loaded to the drive, normally each manufacturer implements a set of "default" homing functions but you need to check the configuration of your drive, especially the Homing Methods... take a look at the documentation of your driver.

Finally, the main purpose of the **BaseNode402** class is to be extended and used as base to implement the extra functionality that you need for the type of control you are trying to do (e.g., Position, Velocity, Torque).

I hope this helps you achieve what you want ;)

af-silva, 2019-05-06:
Hi @Wez50,

If you are satisfied and don't need more help, please close the ticket so this thread can be clean, since this is not a software issue, thanks ;)

Wez50, 2019-06-27:
Very sorry for the delay in closing this, thanks for all the help and best regards. Wess

closed 2019-06-27 by Wez50