canopen-python/canopen #153: when I try to halt the motor ,and later continue the operation ,I got a error. I still have no idea.

Jiongyu, 2019-04-06:
### **error:**

> Traceback (most recent call last):
>   File "/home/tan/ros/module_robot/src/birl_module_robot/UI/script/climbot5d_transmit_jonit_data.py", line 427, in pause_run
>     self.I1.continue_run()
>   File "/home/tan/ros/module_robot/src/birl_module_robot/canopen_communication/modular/modular_I100.py", line 169, in continue_run
>     self.node.sdo[0x6040].bits[8] = 0
>   File "/home/tan/.local/lib/python2.7/site-packages/canopen/variable.py", line 182, in __setitem__
>     self.write()
>   File "/home/tan/.local/lib/python2.7/site-packages/canopen/variable.py", line 194, in write
>     self.variable.raw = self.raw
>   File "/home/tan/.local/lib/python2.7/site-packages/canopen/variable.py", line 85, in raw
>     self.data = self.od.encode_raw(value)
>   File "/home/tan/.local/lib/python2.7/site-packages/canopen/variable.py", line 37, in data
>     self.set_data(data)
>   File "/home/tan/.local/lib/python2.7/site-packages/canopen/sdo/base.py", line 92, in set_data
>     self.sdo_node.download(self.od.index, self.od.subindex, data, force_segment)
>   File "/home/tan/.local/lib/python2.7/site-packages/canopen/sdo/client.py", line 157, in download
>     fp.close()
>   File "/home/tan/.local/lib/python2.7/site-packages/canopen/sdo/client.py", line 387, in write
>     "Unexpected response 0x%02X" % res_command)
> canopen.sdo.exceptions.SdoCommunicationError: Unexpected response 0x43

### **my halt motor and continue operation code:**
```
    def pause_run(self):
        self.node.sdo[0x6040].bits[8] = 1
```
```
    def continue_run(self):
        self.node.sdo[0x6040].bits[0] = 1
        self.node.sdo[0x6040].bits[1] = 1
        self.node.sdo[0x6040].bits[2] = 1
        self.node.sdo[0x6040].bits[3] = 1
        self.node.sdo[0x6040].bits[8] = 0
```

af-silva, 2019-05-06:
Hi Jiongyu, 

Can you please sniff the canbus netowrk using the CanDump utility from [here](https://elinux.org/Can-utils) to see the response message you are receiving. You can refer to this Question #152 to see an example, and also see the state machine of the control node, are you using the DS402 profile??

christiansandberg, 2019-05-06:
If you want to set multiple bits it more efficient to do it yourself. Otherwise each bit operation will result in one upload and one download, i.e. 10 SDO operations in your case. It might cause with problems with the SDO server or CAN traffic.

```python
def continue_run(self):
    val = self.node.sdo[0x6040].raw
    val |= 0xF
    val &= 0x7F
    self.node.sdo[0x6040].raw = val
```

af-silva, 2019-10-16:
@Jiongyu your problem is still happening?

af-silva, 2019-10-23:
Closing the issue, if needed I can open it again. Thanks

closed 2019-10-23 by af-silva