canopen-python/canopen #154: Remote and Local nodes as NmtSlave and NmtMaster

brytonc, 2019-04-10:
I noticed that RemoteNode is defaulted as an, and subsequently tied to, NmtMaster. LocalNode has the same relationship with NmtSlave. As I understand, that ties the two to being a master or slave node respectively.

Would it make more sense to provide a LocalSlaveNode, LocalMasterNode, RemoteMasterNode, and RemoteSlaveNode? I think that would allow for more simulation out-of-the-box. Each user is certainly able to make their own classes, but since 2 of the 4 options are provided, it seems that the other 2 should be as well or am I not understanding the intention?

christiansandberg, 2019-04-11:
Could you give an example of a use case? The RemoteNode is used to control another real node (I.e. it is a master) while LocalNode is meant to be controlled by another master.

brytonc, 2019-04-11:
An example would be if you have several real (hardware exists) slave nodes that you want to control with a virtual (on the computer) master node, you would want to use a local node to act as the master node, right?

christiansandberg, 2019-04-11:
The master does not have to be a node itself unless you have some other master that should talk to your computer. It sounds like you only need RemoteNode.

brytonc, 2019-04-11:
I am confused. In my current scenario, I want to simulate a microcontroller which will be the master on my can bus. To simulate it, I will create a LocalNode on my computer. If I use LocalNode though, it can't be a master, since LocalNode uses NMTSlave.

Am I wrong to use LocalNode in this scenario? Should I be using RemoteNode even though the node I am creating is local to the PC?

christiansandberg, 2019-04-12:
Yes, you don’t need to create a LocalNode unless you want to simulate a slave device with its own node ID.

A master here is just a Python application which creates one or more RemoteNode instances to control the slaves.

brytonc, 2019-04-12:
According to contributor @acolomb in response to another question though:

> A remote node is the controller you want to talk to, which is not your PC where python-canopen is running. A local node would be this library emulating all the state and behavior just as any other CANopen device would behave.

The nomenclature seems off if a RemoteNode is to represent a locally-simulated node.

In the scenario below, would this mapping not be accurate:
Sensor slave -> LocalNode
Master Node -> LocalNode
Motor Slave -> RemoteNode
Encoder Slave -> RemoteNode

[image]

Because of the inherited NMT type, however, such a setup would not function (If I understand correctly).

My earlier proposition would make the setup look like:
Sensor slave -> LocalSlaveNode
Master Node -> LocalMasterNode
Motor Slave -> RemoteSlaveNode
Encoder Slave -> RemoteSlaveNode

I am more than happy to do a PR if this is the case.

christiansandberg, 2019-04-12:
It is correct except that Master Node doesn’t need to be instantiated by some class. You create the code yourself which uses the other Remote and Local node instances to control them.

If you really want the master node to be present on the bus as a slave node (e.g. for some service tool to configure it via SDO) then you can certainly create a LocalNode for that purpose, but it is not needed to act as a master.

brytonc, 2019-04-12:
I do not want the master node to be on the bus as a slave node. On a real (all hardware) can bus, the master is a physical node on the bus. When properly simulating, it makes sense, to me, that it be a node (master, not slave) on the virtual bus as well, but I feel like we are veering off-topic.

My original point is that the provided LocalNode and RemoteNode classes do not cover all potential simulation cases, as I've shown above. A RemoteNode is not necessarily a master. A remote (hardware) node that is a slave does not have an available option; nor does a local (python) node that is a master.

Maybe this is a semantic issue, but my thoughts boil down to:
RemoteNode (a node that is hardware) should be able to be master or slave, and
LocalNode (a node that exists in python) should be able to be master or slave.
Both should connect to the can bus as a node just as physical nodes connect to the can bus.

christiansandberg, 2019-04-12:
The names of the classes may be confusing. The NmtMaster class controls and monitors the state of one particular slave node. A special case is node ID 0 which will be received by all slaves nodes. So a master application just needs one NmtMaster for each node it wants to control, or more conveniently, the RemoteNode which contains it.

The LocalNode and related classes is very new and has never been needed to implement a master.

brytonc, 2019-04-12:
I think there is a disconnect between what I'm trying to convey and what you are interpreting.

I WANT a RemoteNode to be a slave, but the RemoteNode in the library is a master. I am proposing that RemoteNode should actually be two classes: RemoteSlaveNode and RemoteMasterNode.

christiansandberg, 2019-04-12:
Seems like we're going in circles. :) If we look at your example, the motor and encoder nodes will be represented in your master application as RemoteNode since they are "external". When your application writes e.g. `motor.nmt.state = 'OPERATIONAL'` it is acting as a master controlling the slave. The sensor node will be a LocalNode as it will be implemented by yourself in Python.

Now the master you can implement any way that makes sense for your application. It can be a very basic Python script like in the README or docs examples or it can be more sophisticated. For example:

```python
import canopen

network = canopen.Network()

motor = canopen.RemoteNode(5, 'motor.eds')
encoder = canopen.RemoteNode(6, 'encoder.eds')
sensor = canopen.LocalNode(7, 'sensor.eds')
network.add_node(motor)
network.add_node(encoder)
network.add_node(sensor)

network.connect()

# Master application goes here
motor.nmt.state = 'OPERATIONAL'
encoder.nmt.state = 'OPERATIONAL'
sensor.nmt.state = 'OPERATIONAL'

network.disconnect()
```

When the script is executed on your PC it *is* the actual master node. It sends out NMT commands, listens to heartbeat, reads and writes parameters using SDO or whatever you need to do.

That being said, you can still use the individual classes to combine your own node class if you must. You just need to keep track of subscriptions and network associations etc.

acolomb, 2019-04-12:
Let me chime in here, with a more detailed version of what I was trying to explain in the quote above.

The remote nodes (motor, encoder) are represented by objects of class RemoteNode in your program. The MCU simulation is your program, which might be implemented as an instantiated object of your logic written as a class, however there is no such thing predefined in python-canopen. What would your LocalMasterNode class sensibly contain in the library?

Now the NmtMaster class is not a node representation, but rather a collection of functions related to the commands an NMT master may give to an NMT slave node. It's instantiated as a member of each RemoteNode object, meaning that it offers you commands a master may give to that specific node. Your code calling those commands is the actual NMT master in that specific time frame.

Note that master and slave roles are explicitly defined in CANopen for different protocols / message exchanges. The nodes themselves are not inherently master or slave. That is why the library just distinguishes between local and remote.

I implemented for example an MCU (in C, not Python) which controls other nodes, acting as an NMT master. However, it can be told to shut up, switch to operational or restart via NMT commands, acting as a slave and limiting its allowed active communication objects, just like all the other nodes. You could do the same with this library, but your code needs to instantiate an NmtSlave object (I guess) and take care of stopping PDO transmission when switched to pre-operational state for example.

What would your suggested RemoteMasterNode class behave like, since it is not even used in your own example?

acolomb, 2019-04-12:
So to summarize: In the NMT protocol, the "master" is not necessarily a node, but just a role. It's whatever device that drops the appropriate NMT object on the bus.

brytonc, 2019-04-15:
I talked with some coworkers and read more through the code. I made assumptions based on the names and misunderstood the intended purpose of the RemoteNode. Thanks for your help and effort

closed 2019-04-15 by brytonc