canopen-python/canopen #155: Error in node.sdo.upload

MakerSeven, 2019-04-16:
Hello there,
I've just started to use canopen. I have a device (a drive controller for a BLDC motor) declared to be canopen. However so far I've got no EDS or Device configuration file. I have an extensive set of documentation for this device though. So, without proper EDS or Device configuration file I guess I can only use node.sdo.download. I did that  successfully. I have issues with  node.sdo.upload. It doesn't work properly. Sometimes it reads incoming data OK, sometimes not. For instance I use node.sdo.upload(0x3762, 0x00) which should read the actual position (device parameter data 3000h-5FFFh are vendor specific). The values in the object dictionary for device parameter 3762.00 are between -7FFFFFFF and +7FFFFFFF. 
I can read them from candump, for instance:   can0  581   [8]  43 62 37 00 01 00 00 00. This value is properly interpreted by node.sdo.upload as b'\x01\x00\x00\x00'. However then the next read   can0  581   [8]  43 62 37 00 95 26 01 00 is interpreted by node.sdo.upload as b'\x95&\x01\x00' which of course is wrong and doesn't make sense. I could trace the error to the following line in client.py:
SDO_STRUCT.pack_into(request, 0, REQUEST_UPLOAD, index, subindex), but then the debugger faltered and with my humble knowledge of python I thought it doesn't make much sense to go beyond this.
Please advice.

christiansandberg, 2019-04-16:
Its just the printed representation by Python. The character “&” is ASCII code 0x26.

MakerSeven, 2019-04-17:
Hi Christian,
thank you for your quick resolution. I was puzzled by the python representation because bitwise operator didn't accept such a byte string variable.

closed 2019-04-17 by MakerSeven