canopen-python/canopen #159: PDO message not receiving

fabio-mat, 2019-07-30:
Hi, 

Thanks a lot for the canopen library. It's my first contact with can and canopen. So maybe I have missed something fundamental of canopen.
I have some troubles in receiving a pdo message. From my remote_node, i get the heartbeat (every 200 ms) and three pdo's (every 1000ms).

If I use `network.subscribe(0x187,received0x187),` i can catch the tpdo 1 from node 7. But with `node.tpdo[1].wait_for_reception` i catch nothing.
I also get an builtins.KeyError, if a want to read the data of the pdo with `node.tpdo[1]['Battery_Voltage'].phys.`

This is my code:

> import can
> import canopen
> import logging
> import time
> import binascii`
> 
> logging.basicConfig(level=logging.DEBUG)
> 
> network = canopen.Network()
> node = canopen.RemoteNode(7, 'EmulFCA_port.eds', False)
> network.add_node(node)
> 
> network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate=125000)
> def received0x187(identifier, data, unknown):
>       print('identifier %s received with data:' % identifier)
>       print(binascii.hexlify(data))
> 
> network.subscribe(0x187, received0x187)
> 
> node.tpdo[1].wait_for_reception(timeout = 3)
> ChargingTimePhys = node.tpdo[1]['Battery_Voltage'].phys
> 
> network.sync.stop()
> network.disconnect()

I get this in dhe Debug-I/O Console:

> DEBUG:canopen.pdo:TPDO Map as 1
> DEBUG:canopen.pdo:RPDO Map as 0
> DEBUG:can:can config: {'bitrate': 125000, 'channel': 'PCAN_USBBUS1', 'interface': 'pcan'}
> INFO:canopen.network:Connected to 'PCAN_USBBUS1'
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> identifier 391 received with data:
> b'0200a861c4091701'
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> identifier 391 received with data:
> b'0200a861c4091701'
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> identifier 391 received with data:
> b'0200a861c4091701'
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5
> DEBUG:canopen.nmt:Received heartbeat can-id 1799, state is 5

And this in my Exceptions-Console:

> File "C:\Users\fabio.mathis\AppData\Local\Continuum\anaconda3\Lib\site-packages\canopen\pdo\base.py", line 215, in __getitem__
>   var = self.__getitem_by_index(int(key, 16))
> 
> builtins.ValueError: invalid literal for int() with base 16: 'Battery_Voltage'
> 
> During handling of the above exception, another exception occurred:
> 
> File "C:\PythonWorkspace\canOpen.py", line 35, in <module>
>   ChargingTimePhys = node.tpdo[1]['Battery_Voltage'].phys
> File "C:\Users\fabio.mathis\AppData\Local\Continuum\anaconda3\Lib\site-packages\canopen\pdo\base.py", line 217, in __getitem__
>   var = self.__getitem_by_name(key)
> File "C:\Users\fabio.mathis\AppData\Local\Continuum\anaconda3\Lib\site-packages\canopen\pdo\base.py", line 203, in __getitem_by_name
>   value, ', '.join(valid_values)))
> 
> builtins.KeyError: 'Battery_Voltage not found in map. Valid entries are '
> 
> 
And this is my eds file, which a saved as txt to drag here:

[EmulFCA_port.txt](https://github.com/christiansandberg/canopen/files/3447143/EmulFCA_port.txt)

And this my log with PCAN:
[image]

christiansandberg, 2019-07-30:
You also need to read the PDO configuration, otherwise it doesn’t know what the PDO is or what it contains.

node.tpdo[1].read()

fabio-mat, 2019-07-31:
Thanks for the fast response. 

Unfortenately, I havent implementend SDO's yet (on my microcontroller-node with id 7). So, I have to do this first...

With `node.tpdo[1].read()` the library wants to read the pdo information with of the node with sdo commands, is this correct? I thougt the libary gets the information about the pdo from the eds file...
Is it not implemented, or is it not possible to get all the necessary information about the pdo from the eds file (if I have static/strictly pre-defined mapping)? Which information gets the library from the eds file?

fabio-mat, 2019-07-31:
And is there a way to set the pdo configuration with the current library without using sdo's?

christiansandberg, 2019-08-02:
Quite limited right now I'm afraid. But it is possible with the current set of API methods and attributes. This might work:

```python
pdo = node.rpdo[1]
pdo.cob_id = 0x187
pdo.clear()
pdo.add_variable('Battery_Voltage')
pdo.enabled = True
network.subscribe(pdo.cob_id, pdo.on_message)
```

fabio-mat, 2019-08-07:
Thanks a lot for your help.
Your suggestion works.

Are you planning to implement reading pdo mapping from eds file in the future?

christiansandberg, 2019-08-12:
I doubt that I will have time to do that but I would happily accept a pull request!

closed 2019-08-12 by christiansandberg