canopen-python/canopen #163: Emergency message not received

ChristopheMarques, 2019-09-24:
Hi, 
I have a communication sequence where, sometimes, Emergency message seems not received; timeout occurs, but it is logged in CAN traces made with Bosch BusMaster (canopen 0.9.0 version). 
Have you got some tips/workaround? Thanks in advance.

## CAN TRACE ##
17:02:30:4711 Rx 1 0x27F s 8 20 F4 6C 11 00 20 01 00 
17:02:30:4711 Rx 1 0x1FF s 3 20 F4 00 
17:02:30:4721 Rx 1 0x0A0 s 8 01 00 81 00 01 00 00 00 
17:02:30:4761 Rx 1 0x27F s 7 20 F0 44 4D 44 31 34 
17:02:30:4761 Rx 1 0x1FF s 3 20 F0 00 
*17:02:30:4871 Rx 1 0x27F s 8 20 F4 72 11 00 20 10 00 -> 0045.967-basicStandardProtocol-MSG Set Warning code 0x10*
*17:02:30:4871 Rx 1 0x1FF s 3 20 F4 00                -> Response*
*17:02:30:4891 Rx 1 0x0A0 s 8 10 00 81 00 10 00 00 00 -> EMCY not received... -> It's not in the log -> Timeout occurs -> "P" is displayed*
*17:02:31:3601 Rx 1 0x720 s 1 7F                      -> DEBUG:canopen.nmt:Received heartbeat can-id 1824, state is 127*
17:02:32:3601 Rx 1 0x720 s 1 7F 
17:02:33:3601 Rx 1 0x720 s 1 7F 
17:02:34:3591 Rx 1 0x720 s 1 7F 
17:02:35:3591 Rx 1 0x720 s 1 7F 

## Code ##
```
self.pcMST.writeVariable(0x20, self.pcMST.varListOut['DriveSupervise_WarningsU16'], wrn)
        
if(wrn):
    self.message(f"Set Warning code 0x{wrn:02x}")
else:
    self.message("Reset warning")

try:
    tmp = self.node.emcy.wait(timeout=1)
    self.check_err(tmp.register,err_reg_to_check, wrn)
except:
    print("P")
    exit()
```

## LOG ##
DEBUG:canopen.variable:Writing MDModeSelector (0x3000:0) = 1
0045.934-basicStandardProtocol-MSG Set Error code 0x01
INFO:canopen.emcy:Got Code 0x0001, Error Reset / No Error
0045.955-basicStandardProtocol-INF basicStandardProtocol_Error reg: 0x81=OK
0045.967-basicStandardProtocol-MSG Set Warning code 0x10
DEBUG:canopen.nmt:Received heartbeat can-id 1824, state is 127
P

af-silva, 2019-10-03:
Hi, 

Without more code context is difficult to pinpoint the problem, but in the snippet you provided the only part of the code related to the canopen api is the ``` self.node.emcy.wait(timeout=1) ``` ...
I don't think this is the best way to implement the mechanism to receive emergency messages from the canbus node ...

I don't know your requirements or your end goal, but for me the best approach is first to configure the emcy to listen on incoming messages and don't use the 1 second timeout with the wait. 

if you open the [remote.py](https://github.com/christiansandberg/canopen/blob/master/canopen/node/remote.py) in the mesthod ```associate_network(self, network):``` configures the node to subscribe to the network on new emcy messages. 
```
...
network.subscribe(0x700 + self.id, self.nmt.on_heartbeat)
network.subscribe(0x80 + self.id, self.emcy.on_emcy)
network.subscribe(0, self.nmt.on_command)
...
```
I would suggest to use the remote node implementation and then configure the callback for incoming emcy messages, instead of blocking the execution and waiting 1 second at a time, since you can be loosing messages in between.

part of implementation of a base class extending the BaseNode402 class: `class BaseNode(BaseNode402):`
```python
...

self.emcy.add_callback(self.handle_emcy)

...

def handle_emcy(self, msg):
    """
    :param msg:
    """
    # This is a special case for the safe torque off activated (emergency button)
    if msg.code == 0xFF60:
          # do something here with the message
...

af-silva, 2019-10-23:
@ChristopheMarques hi, since you didn't reapond in the last 20 day I'm closing this issue. If you need I can open it again. Thanks

closed 2019-10-23 by af-silva