canopen-python/canopen #164: CANopen configuration

CarlDebrouwere, 2019-10-11:
Hello, I am quite new in the world of coding.
For my work I have to make a CANopen network for controlling 4 linear actuators, which are connected in a serial way.
I managed to make a CANopen network via:
network = canopen.Network()

But I am kinda stuck here, I don't really know how to address to these nodes.
Could somebody help me with this?

af-silva, 2019-10-15:
Hi,

Can BUS is normally composed by two wires CAN Hi and CAN Lo (image below)

[image]
[Courtesy of Sparkfun](https://learn.sparkfun.com)

So yes, if you have them connected this way you are correct, they all share the same bus of communication. 
Now, the next step is to know the protocol supported by the controllers of the actuators, if it is CAN DS402 you can use the BaseNode402 base class provided by the project, you can find it in the profiles folder. If not, if the manufacturer uses another implementation of the CAN protocol you can extend the  RemoteNode and add the necessary functionality. 

Since you say you are new at programing this tasks can be somehow daunting, so use this check list to move forward: 

- First and foremost get an introduction to the CAN Bus network, you can read [this wiki page](https://en.wikipedia.org/wiki/CAN_bus);
- Which CAN protocol is supported by the drivers? (e.g., DS 402);
- The manufacturer provides the EDS files ? If so download them;
- Look at the exemple provided in the README.rst and try to understand what is happening, the code is well documented but you need to know some bases of how CAN networks work to understand it;
- If your drives support the DS 402 profile you can look at the BaseNode402 and look at the code, it implements the state machine used by this profile to monitor the connection and state of each connected controller;

Try going forward from here and if you have a more concrete doubt I will try to help. 

best regards

CarlDebrouwere, 2019-10-21:
Dear,

I managed to get the EDS files, I wasn't able to find out which CAN  
protocil is supported by the controllers.

Regards,
Carl

CarlDebrouwere, 2019-10-22:
It is indeed DSP 402.
But where can I find this porfiles folder?

Regards,
Carl

af-silva, 2019-10-22:
Hi Carl,

The profiles folder is at <project root>/canopen/profiles 

If you want to get an idea of how to implement a simple node based on the DS402 profile please refer to this question #152

closed 2020-05-15 by af-silva