canopen-python/canopen #165: [Q] Simulate nodes with canopen lib

ngblume, 2019-11-01:
Hey,

I'm starting with using this library and for testing purposes I would like to communicate with a "software" node on the same computer.
I found #127, which states:
"A remote node is the controller you want to talk to, which is not your PC where python-canopen is running. A local node would be this library emulating all the state and behavior just as any other CANopen device would behave. So others can treat your program like you see the Curtis controller."
That suggests for me, that simulating node is possible and I would simply add a local node to my network with a proper eds file and hen talk to that node...

Is this correct ??

What confuses me, is that #47, which suggests that SDO server already works when simulating nodes (local nodes) , but not all functionalities of a CANopen slave node..
Is that correct ?

And last, but not least, is there an example with a local node, a simple eds with just one entry and a simple sdo upload (=read) ?
Would be greatly appreciated..

Thanks !

Cheers
Niels

christiansandberg, 2019-11-01:
Maybe you can look at test/test_local.py to get an idea of how it works and what is supported.

ngblume, 2019-11-02:
Hey,

thanks, I checked it out and it made my idea working...

Except block transfer....
The terminal shows "Block download is not supported"..
When looking at the examples I see the firmware download (write to a node) example:
https://canopen.readthedocs.io/en/latest/sdo.html 
It shows block transfer for the download to the node.

Am I running an old version 1.0.0 (installed via pip) or do I need to activate block transfer separately (especially for local nodes) ?

The example I built is attached.
[test-scanner.zip](https://github.com/christiansandberg/canopen/files/3799974/test-scanner.zip)

Thanks !
Cheers
Niels

christiansandberg, 2019-11-02:
Block transfer has not been implemented in the SDO server.

ngblume, 2019-11-02:
Meaning that you can use it with "real" remote nodes, but not with
simulated local nodes...??
Since it is implemented in alpha stage in the SDO client...?

christiansandberg, 2019-11-02:
Yes exactly. You can initiate block transfer as a master to another slave node but not the other way currently.

ngblume, 2019-11-03:
Hey,

upon further testing I extended my example to include using an existing bus instead of creating one by connecting the network to an interface..
The code is attached.
I'm getting the following exception when running the code:

> Traceback (most recent call last):
  File "test-scanner.py", line 70, in <module>
    localSimNode_0x05.nmt.state = 'INITIALISING'
  File "/home/ngblume/.local/lib/python3.6/site-packages/canopen/nmt.py", line 103, in state
    self.send_command(code)
  File "/home/ngblume/.local/lib/python3.6/site-packages/canopen/nmt.py", line 215, in send_command
    self.network.send_message(0x700 + self.id, [0])
  File "/home/ngblume/.local/lib/python3.6/site-packages/canopen/network.py", line 202, in send_message
    self.check()
  File "/home/ngblume/.local/lib/python3.6/site-packages/canopen/network.py", line 247, in check
    exc = self.notifier.exception
AttributeError: 'NoneType' object has no attribute 'exception'

Im a little baffeled as to the reason, since the code works exactly as I expect to work, when using the commented-out connect command and removing the notifier section....

Any ideas what's causing that...?
Thanks !

Cheers Niels

christiansandberg, 2019-11-04:
That is kind of a bug. You could probably go around it by setting `network.notifier = your_notifier`.

ngblume, 2019-11-05:
Could you elaborate a little as to what you mean?
Do I have to develop a function "your_notifier" ?

ngblume, 2019-11-05:
One follow-up:
When I'm using the low-level functions, there is no data conversion.
Returned data is always array of bytes...
I switched to the low-level functions, since I want to write a wrapper function, that receives index and subindex and calls the sdo library function. But passing the index and subindex to the library was not working correctly, the issue being, that reading parameters that have no subindex, cannot be adressed with a subindex.

Any recommendation as to how to implement this most efficiently?

christiansandberg, 2019-11-05:
Maybe something like this:

```python
def get_sdo_object(node, index, subindex):
    obj = node.sdo[index]
    if isinstance(obj, canopen.sdo.Variable):
        return obj
    else:
        return obj[subindex]
```

ngblume, 2019-11-06:
Thanks for the pointer !!

I think I need to redesign my wrapper class from the ground up ..
If I come across other problems / questions, I will post again...

closed 2019-11-06 by ngblume