canopen-python/canopen #166: TX-PDO not being send

ngblume, 2019-11-07:
Hey,

quick follow up question regarding TX-PDOs:
I seem to have correctly configured those and they are enabled, but they are not send by the localNode.
I tried TransMode 1, 254 and 255 with different event_time times and for TransMode=1 also sending SYNC messages (which are send out onto the bus).
This is the code I'm running to set the TX-PDO1 (it has been set via the EDS file before, as you will see in the log)

```
network[node_id].tpdo.read()
network[node_id].tpdo[pdo_number].clear()
network[node_id].tpdo[pdo_number].add_variable(0x6011, 1)
network[node_id].tpdo[pdo_number].add_variable(0x6011, 2)
network[node_id].tpdo[pdo_number].trans_type = 1
network[node_id].tpdo[pdo_number].event_timer = 10
network[node_id].tpdo[pdo_number].enabled = True
# Save new PDO configuration to node
network[node_id].tpdo[pdo_number].save()
```

When enabling DEBUG logging, I get the following trace (combined with logging from my program; I removed additional log entries to keep it short):

```
DEBUG:canopen.pdo:TPDO Map as 2
Addind remote node:  0x3
{3: <canopen.node.remote.RemoteNode object at 0x7fd3d2d49550>}
INFO:canopen.nmt:Changing NMT state from INITIALISING to INITIALISING
DEBUG:canopen.variable:Writing Producer Heartbeat Time (0x1017:0) = 2500
INFO:canopen.nmt:Start the hearbeat timer, interval is 2500 ms
INFO:canopen.nmt:Changing NMT state from PRE-OPERATIONAL to PRE-OPERATIONAL
INFO:canopen.nmt:Sending NMT command 0x80 to node 3
INFO:canopen.nmt:New NMT state PRE-OPERATIONAL, old state INITIALISING
====================================
DEBUG:canopen.sdo.client:Reading 0x1800:1 from node 3
DEBUG:canopen.variable:Value of Transmit PDO Communication Parameter 0.COB ID (0x1800:1) is 387
INFO:canopen.pdo.base:COB-ID is 0x183
INFO:canopen.pdo.base:PDO is enabled
INFO:canopen.pdo.base:RTR is allowed
DEBUG:canopen.sdo.client:Reading 0x1800:2 from node 3
INFO:canopen.pdo.base:Transmission type is 255
INFO:canopen.pdo.base:Inhibit time is set to 500 ms
INFO:canopen.pdo.base:Event timer is set to 780 ms
INFO:canopen.pdo.base:SYNC start value is set to 0 ms
DEBUG:canopen.sdo.client:Reading 0x1A00:0 from node 3
DEBUG:canopen.variable:Value of Transmit PDO Mapping Parameter 0.Number of entries (0x1A00:0) is 3
DEBUG:canopen.sdo.client:Reading 0x1A00:1 from node 3
DEBUG:canopen.variable:Value of Transmit PDO Mapping Parameter 0.PDO Mapping Entry_1 (0x1A00:1) is 1610678536
INFO:canopen.pdo.base:Adding DI.DI-InputGroup1 (0x6001:1) at bits 0 - 7 to PDO map
DEBUG:canopen.variable:Value of Transmit PDO Mapping Parameter 0.PDO Mapping Entry_2 (0x1A00:2) is 268501000
INFO:canopen.pdo.base:Adding Error Register (0x1001:0) at bits 8 - 15 to PDO map
DEBUG:canopen.variable:Value of Transmit PDO Mapping Parameter 0.PDO Mapping Entry_3 (0x1A00:3) is 1611727392
INFO:canopen.pdo.base:Adding AI.AI-Input2 (0x6011:2) at bits 16 - 47 to PDO map
====================================
DEBUG:canopen.sdo.client:Reading 0x1801:1 from node 3
DEBUG:canopen.variable:Value of Transmit PDO Communication Parameter 1.COB ID (0x1801:1) is 643
INFO:canopen.pdo.base:COB-ID is 0x283
INFO:canopen.pdo.base:PDO is enabled
INFO:canopen.pdo.base:RTR is allowed
INFO:canopen.pdo.base:Transmission type is 255
INFO:canopen.pdo.base:Inhibit time is set to 500 ms
INFO:canopen.pdo.base:Event timer is set to 980 ms
INFO:canopen.pdo.base:SYNC start value is set to 0 ms
DEBUG:canopen.sdo.client:Reading 0x1A01:0 from node 3
DEBUG:canopen.variable:Value of Transmit PDO Mapping Parameter 1.Number of entries (0x1A01:0) is 2
DEBUG:canopen.variable:Value of Transmit PDO Mapping Parameter 1.PDO Mapping Entry_1 (0x1A01:1) is 1611727136
DEBUG:canopen.sdo.client:Reading 0x1A01:1 from node 3
DEBUG:canopen.variable:Value of Transmit PDO Mapping Parameter 1.PDO Mapping Entry_1 (0x1A01:1) is 1611727136
INFO:canopen.pdo.base:Adding AI.AI-Input1 (0x6011:1) at bits 0 - 31 to PDO map
DEBUG:canopen.variable:Value of Transmit PDO Mapping Parameter 1.PDO Mapping Entry_2 (0x1A01:2) is 1611727392
INFO:canopen.pdo.base:Adding AI.AI-Input2 (0x6011:2) at bits 32 - 63 to PDO map
INFO:canopen.pdo.base:Adding AI.AI-Input1 (0x6011:1) at bits 0 - 31 to PDO map
INFO:canopen.pdo.base:Adding AI.AI-Input2 (0x6011:2) at bits 32 - 63 to PDO map
INFO:canopen.pdo.base:Setting COB-ID 0x183 and temporarily disabling PDO
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 0.COB ID (0x1800:1) = 2147484035
INFO:canopen.pdo.base:Setting transmission type to 1
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 0.Transmission Type (0x1800:2) = 1
INFO:canopen.pdo.base:Setting inhibit time to 50000 us
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 0.Inhibit Time (0x1800:3) = 500
INFO:canopen.pdo.base:Setting event timer to 10 ms
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 0.Event Timer (0x1800:5) = 10
INFO:canopen.pdo.base:Setting SYNC start value to 0
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 0.SYNC start value (0x1800:6) = 0
DEBUG:canopen.variable:Writing Transmit PDO Mapping Parameter 0.Number of entries (0x1A00:0) = 0
INFO:canopen.pdo.base:Writing AI.AI-Input1 (0x6011:1, 32 bits) to PDO map
DEBUG:canopen.variable:Writing Transmit PDO Mapping Parameter 0.PDO Mapping Entry_1 (0x1A00:1) = 1611727136
INFO:canopen.pdo.base:Writing AI.AI-Input2 (0x6011:2, 32 bits) to PDO map
DEBUG:canopen.variable:Writing Transmit PDO Mapping Parameter 0.PDO Mapping Entry_2 (0x1A00:2) = 1611727392
DEBUG:canopen.variable:Writing Transmit PDO Mapping Parameter 0.Number of entries (0x1A00:0) = 2
INFO:canopen.pdo.base:Enabling PDO
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 0.COB ID (0x1800:1) = 387
INFO:canopen.pdo.base:Setting COB-ID 0x283 and temporarily disabling PDO
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 1.COB ID (0x1801:1) = 2147484291
INFO:canopen.pdo.base:Setting transmission type to 255
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 1.Transmission Type (0x1801:2) = 255
INFO:canopen.pdo.base:Setting inhibit time to 50000 us
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 1.Inhibit Time (0x1801:3) = 500
INFO:canopen.pdo.base:Setting event timer to 980 ms
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 1.Event Timer (0x1801:5) = 980
INFO:canopen.pdo.base:Setting SYNC start value to 0
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 1.SYNC start value (0x1801:6) = 0
DEBUG:canopen.variable:Writing Transmit PDO Mapping Parameter 1.Number of entries (0x1A01:0) = 0
INFO:canopen.pdo.base:Writing AI.AI-Input1 (0x6011:1, 32 bits) to PDO map
DEBUG:canopen.variable:Writing Transmit PDO Mapping Parameter 1.PDO Mapping Entry_1 (0x1A01:1) = 1611727136
INFO:canopen.pdo.base:Writing AI.AI-Input2 (0x6011:2, 32 bits) to PDO map
DEBUG:canopen.variable:Writing Transmit PDO Mapping Parameter 1.PDO Mapping Entry_2 (0x1A01:2) = 1611727392
DEBUG:canopen.variable:Writing Transmit PDO Mapping Parameter 1.Number of entries (0x1A01:0) = 2
INFO:canopen.pdo.base:Enabling PDO
DEBUG:canopen.variable:Writing Transmit PDO Communication Parameter 1.COB ID (0x1801:1) = 643
====================================
INFO:canopen.nmt:Changing NMT state from PRE-OPERATIONAL to OPERATIONAL
INFO:canopen.nmt:Sending NMT command 0x1 to node 3
INFO:canopen.nmt:Node 3 received command 1
INFO:canopen.nmt:New NMT state OPERATIONAL, old state PRE-OPERATIONAL
DEBUG:canopen.nmt:Received heartbeat can-id 1795, state is 5
====================================
Found node:  0x3
DEBUG:can.interfaces.socketcan.socketcan:Sending BCM command
DEBUG:can.interfaces.socketcan.socketcan:Stopping periodic task
```

Any help would be greatly appreciated !

Cheers
Niels

christiansandberg, 2019-11-08:
Sorry, PDO for local nodes have not been implemented yet, it just uses the remote node implementation for now which provides some basic functionality but not all. Most importantly it does not react on parameter changes. You need to call the start() method of each map you want transmitted.

ngblume, 2019-11-08:
Thanks for the info!

Just to clarify...
I can use the start() function, but the sent-out PDOs do not use the values from the LocalNode ??
I got the sending working, but the values from the localNode are ignored.
Could you point me to an example of using simulated nodes ?

christiansandberg, 2019-11-12:
I think if you treat you local node's TPDO like a remote node's RPDO (configuring, setting values etc) then it should be possible to get it sending some values. Follow the examples here: https://canopen.readthedocs.io/en/latest/pdo.html

Setting values through Object Dictionary won't work. The best would be to implement local node PDO but I don't think I'll have the time to do that.

closed 2020-05-15 by af-silva