canopen-python/canopen #167: BaseNode402 Transition from 'OPERATION ENABLED' to 'QUICK STOP ACTIVE'

JulianS-Uni, 2019-11-07:
Hi there,

I have a problem when I do a transition form the state 'OPERATION ENABLED' to 'QUICK STOP ACTIVE'. I always get an `ValueError: 'Illegal transition from 'SWITCH ON ENABLED' to 'QUICK STOP ACTIVE'.` Other transitions are working fine, and besides the error, the Motor is doing the transition just fine. I came to the conclusion, that the problem may be, that the transition to 'QUICK STOP ACTIVE' immediately triggers a transition to 'SWITCH ON ENABLED' so that the motor never really is in state 'QUICK STOP ACTIVE' so that your routine `while self.state != new_state:` never gets true and ends up comparing the current state which is 'SWITCH ON ENABLED' with 'QUICK STOP ACTIVE' which then is an illegal transition.

Do you have an idea how to fix this?

af-silva, 2019-11-07:
Hi @JulianS-Uni,

That is a strange behavior. I will look into it, but in the meanwhile at what time `ValueError: 'Illegal transition from 'SWITCH ON ENABLED' to 'QUICK STOP ACTIVE'.` this appears ??? 

In the file [p402.py](https://github.com/christiansandberg/canopen/blob/master/canopen/profiles/p402.py) you can find the state transition table and there is no **_"SWITCH ON ENABLED"_** command or state defined. 

Here is the CiA 402 Power State Machine:
[image]
(image from [Nanotec](https://en.nanotec.com/products/manual/PD2C_CAN_EN/general%252Fds402_power_state_machine.html/))

Can you please confirm what the StatusWord holds when this error happens.

Thanks.

JulianS-Uni, 2019-11-07:
Sorry, my bad. It is not SWITCH ON ENABLED it is SWITCH ON DISABLED.

af-silva, 2019-11-09:
No problem, yes, but the correct transition is from Quick Stop Active -> Switch on Disabled and the error you have is a state change the other way around that is "illegal". That is way I said it is a strange behavior ... but like you said your drive is changing automatically from Quick Stop Active to Switch on Disabled ... so, my guess is the following, and I would appreciate if you could validate this since now I don't have direct and easy access to a working controller, please edit the p402.py file and edit the following line `
        ('OPERATION ENABLED', 'QUICK STOP ACTIVE'):       CW_QUICK_STOP,  # transition 11` and replace the `CW_QUICK_STOP` with `CW_SWITCH_ON_DISABLED` and check if the error still happens. I will wait your feedback ;)

JulianS-Uni, 2019-11-09:
>  please edit the p402.py file and edit the following line ('OPERATION ENABLED', 'QUICK STOP ACTIVE'): CW_QUICK_STOP, # transition 11 and replace the CW_QUICK_STOP with CW_SWITCH_ON_DISABLED and check if the error still happens

I will test on monday. But i am pretty sure it will work, although it will trigger transition 9 (Operation Enabled -> Switch on Disabled) and not transition 11 (Operation Enabled -> Quickstop). When my motor is in Operation Enabled and i set its state to Quick Stop Active your code gets the right CW_Code (i think its 2) and sends it. Your code then gets into a while loop checking if the desired state is reached:
 ``` Python
while self.state != next_state:
                    if time.time() > t_to_next_state:
                        break
                    time.sleep(0.01)  # 10 milliseconds of sleep
```
Meanwhile my motor gets the controlword and changes to state Quick Stop Active which triggers it to send its statusword back over PDO (configured so that it sends it statusword each time it gets changed). I am asuming that my motor sends the statusword twice, first time sending status Quick Stop Active and immeditly after Switch On Disabled because of the automatic state change. All this probably happens while your code is in the while loop in the 10 millis sleep phase. So when leaving sleep and the while loop checks again for `self.state != next_state` self.state is then Switch on Disabled already, so the check would be `'SWITCH ON DISABLED' != 'QUICK STOP ACTIVE'` . So that the inner while loop wont ever end and runs into its timeout. So after that, the outer while loop gets executed again and checks for `self.state != new_state` ,  which still is `'SWITCH ON DISABLED' != 'QUICK STOP ACTIVE'`. So the while gets executed and it checks for:
``` Python
 # get the code from the transition table
                code = State402.TRANSITIONTABLE[ (self.state, next_state) ]
```
which will give us the ValueError since the illegal transition.

af-silva, 2019-11-09:
Yeah, you are right. OK then you can make two tests:

1st: Change the timeout to a even lower value, I guess 0.001 (just for the processor to breath a little), and check it's behavior. (you are receiving messages at state changes right?, you are not using the synchronization mechanism??? ).

2nd: We can add a second verification for the state transition 11 and check both StatusWord's for the `QUICK STOP ACTIVE` and `SWITCH ON DISABLED`  when a `QUICK STOP ACTIVE` transition is requested and and update the `next_state ` variable.

JulianS-Uni, 2019-11-09:
> (you are receiving messages at state changes right?, you are not using the synchronization mechanism??? ). 

Yes right, so your suggestion could work. I will try this.

> We can add a second verification for the state transition 11

Yes, thats what i wanted to express with my code suggestion above. But i am sure there is a cleaner solution than my code snippet.

JulianS-Uni, 2019-11-11:
Okay i tried setting the sleep to 0.0001 and still get the error. But i checked the candump:
```
  can0  201   [2]  02 00
  can0  181   [2]  17 16
  can0  181   [2]  40 06
  can0  181   [2]  40 02
```
First entry is the controlword setting the quick stop.
Second is the statusword from the motor 17 equals to 0001 0111 which equals to quickstop (x00x 0111)
Third is the second statusword message with 40 equals to 0100 0000 which eqals to switch on disabled (x1xx 0000)
Not quite sure about the forth message.

So i think it would be better to go with your 2nd solution.

af-silva, 2019-11-13:
Sorry @JulianS-Uni for the late response. 

You can try your solution ;)

```python
@state.setter
    def state(self, new_state):
        """ Defines the state for the DS402 state machine
        :param string new_state: Target state
        :param int timeout:
        :raise RuntimeError: Occurs when the time defined to change the state is reached
        :raise TypeError: Occurs when trying to execute a ilegal transition in the sate machine
        """
        t_to_new_state = time.time() + 8  # 800 milliseconds tiemout
        while self.state != new_state:
            try:
                if new_state == 'OPERATION ENABLED':
                    next_state = State402.next_state_for_enabling(self.state)
                else:
                    next_state = new_state
                # get the code from the transition table
                code = State402.TRANSITIONTABLE[ (self.state, next_state) ]
                # set the control word
                self.controlword = code
                # timeout of 400 milliseconds to try set the next state
                t_to_next_state = time.time() + 0.4
                while self.state != next_state:
                    if time.time() > t_to_next_state:
		        # Special case! The transition to QUICK STOP ACTIVE triggers the automatic switch on the driver to SWITCH ON DISABLED
			if next_state == 'QUICK STOP ACTIVE' and self.state == 'SWITCH ON DISABLED':  
			    new_state = self.state
                        break
                    time.sleep(0.0001)  # 10 milliseconds of sleep
            except KeyError:
                raise ValueError('Illegal transition from {f} to {t}'.format(f=self.state, t=new_state))
            # check the timeout
            if time.time() > t_to_new_state:
                raise RuntimeError('Timeout when trying to change state')
            time.sleep(0.01)  # 10 miliseconds of sleep

```

We can also force the `new_state = self.state` to be the `SWITCH ON DISABLED` instead of getting the variable from the `self.state`. This is a special case, I think, if the drivers switch automatically from `QUICK STOP ACTIVE` to `SWITCH ON DISABLED`.

Choose and try the solution you find more appealing and if possible test it, if it works you can make a Pull Request to improve the API.

If this behavior (auto switch from `QUICK STOP ACTIVE` to `SWITCH ON DISABLED`) is not typical or standard we should also add a flag for this "workaround".

What do you think??

gsulc, 2019-11-20:
I'd like to chime in on this. Is there any indication that this automatic transition is prevalent among manufactured drives? This seems like non-standard behavior to me, which I think the library should not support, unless the problem lies in not correctly getting the current state.

JulianS-Uni, 2019-11-21:
@gsulc Yes this is standard behaviour. Look for object 605A for more information. Most devices seem to support code 1 and 2 which do this automatic transition. Some devices like mine as well also support code 5 and 6 which keeps the device in Quick Stop. So the best solution would make use of the current configuration of object 605A.

gsulc, 2019-11-21:
@JulianS-Uni: Hm.. code 0 seems to define this behavior, but I'm not seeing it for 1 and 2. Using the option code sounds like a good idea. One would hope that if the option code object is unused, that the drive wouldn't transition automatically.

JulianS-Uni, 2019-11-21:
@gsulc 
> Hm.. code 0 seems to define this behavior, but I'm not seeing it for 1 and 2.

Object Index 605A Sub-Index 00 :
_[CANopen_HB_engl.pdf](https://www.engelantriebe.de/pdf/CANopen_HB_engl.pdf) search for 605A_
> Specifies the behaviour of the device, when it is switched to the state Quick Stop Active by the master.

> In state Quick Stop Active, the drive decelerates speed controlled as specified in Quick Stop Option Code until stop **and switches to the state Switch On Disabled** (power stage disabled).

Her is another [AKD CANopen Kommunikationshandbuch DE (REV U).pdf](https://www.kollmorgen.com/sites/default/files/public_downloads/AKD%20CANopen%20Kommunikationshandbuch%20DE%20%28REV%20U%29.pdf) it is german but you can copy paste it to google translator.
Bit:

> 1  --- Bremsen mit Verzögerungsrampe und **Übergang in "Einschalten Disabled**"
> 2  --- Bremsen mit Schnellhaltrampe und Übergang in "Einschalten Disabled"
> 5  --- Bremsen mit Verzögerungsrampe und **in "Schnellhalt Aktiv" bleiben**
> 6  --- Bremsen mit Schnellhaltrampe und in "Schnellhalt Aktiv" bleiben

It is even in here [Nanotec](https://en.nanotec.com/products/manual/PD2C_CAN_EN/general%252Fds402_power_state_machine.html/) which we linked already.

closed 2020-04-19 by af-silva