canopen-python/canopen #168: Status word is not configured in this device

gsulc, 2019-11-12:
During setup of the 402 state machine, it is easy to produce the error message, "The object 0x6041 (Statusword) is not configured in this device."  This is because the `statusword` getter returns a value from the `tpdo_values` dictionary, which the `state` getter uses, which the `state` setter uses.

It seems the only way `setup_402_state_machine()` would ever work is if there is a TPDO set up to transmit on SYNC and a SYNC message is sent out. Typically, the Statusword is set up to asynchronously transmit on an event.

A minimum patch would be to change the `statusword` getter to `try` using the `tpdo_values` dictionary and then use an SDO to get the Statusword if it fails.

af-silva, 2019-11-13:
Hi @gsulc ,

You can add the fallback to the SDO mechanism in the StatusWord property 

```python

    @property
    def statusword(self):
        """Returns the last read value of the Statusword (0x6041) from the device.
        :raise ValueError: The Object 0x6041 (Statusword) is not configured in this device.
        """
        try:
            return self.tpdo_values[0x6041]
        except KeyError:
            logger.warning('The object 0x6041 is not supported by the PDO mechanism, fallback to SDO')
            return self.sdo[0x6041].raw = value
```

I have implemented a similar fallback mechanism for the ControlWord. In this case I dind't use it because it adds to much latency in the communication (SDO communications it's much slower than PDO). 

What do you think, this solves your issue???

Thanks.

gsulc, 2019-11-14:
Of course, the latency will be bad with SDO compared to PDO. This would ameliorate the issue, and would is a good idea in general. It is a fallback, so the "usual" performance is good, and it would make the method behave in a more clear way (as an api consumer). As a user, I would not expect this method to fail if I don't have it setup as an RPDO.

The way I started to hack around it in my own code was to introduce my own pre-init step in which I set: 
```python
node.tpdo_values[0x6041] = node.sdo[0x6041].raw
```

But I'm starting to think there was something up with the boards I was using. Because things seem to work fine on a different set of boards, and I all the RPDOs are set to a transmission type of 255. So I don't clearly understand the root cause yet.

I was thinking it may also be a good idea to populate the cached tpdo_values dictionary using SDOs as part of the initialization process.

af-silva, 2019-11-14:
Hi @gsulc , pushed a code update.
Appreciate your feedback. 

Regarding this:
> I was thinking it may also be a good idea to populate the cached tpdo_values dictionary using SDOs as part of the initialization process.

If you want you can tackle this and make a PR and contribute in improving this API. If you want, please kindly open a new issue with this in mind for further discussion.

For now, i will assume this issue closed.

Thanks.

closed 2019-11-14 by af-silva

gsulc, 2019-11-14:
Sure. I will investigate a bit more to be sure about the issue before posting.

commit 1a21371c (2019-11-14): added fallback in statusword property, resolves #168

Enable the statusword to get the value from the driver even if it's not configured in the TPDOS through the SDO mechanism.