canopen-python/canopen #169: DS402 State Transition 'QUICK STOP ACTIVE' to 'OPERATION ENABLED'

gsulc, 2019-11-14:
The DS 402 specification indicates the state transition 16, 'QUICK STOP ACTIVE' to 'OPERATION ENABLED', is optional. Some devices do not support it. This software library is the perfect place to handle this: try the transition and if it doesn't work, fall through to the next-best thing: go to 'SWITCH ON DISABLED' and then 'OPERATION ENABLED'. Bonus points: remember this for the node and don't try it again the next time. I can take this one.

I monkey-patched the method below, and it works fine:
```python
def enable_operation(self):
    state = self.state

    if state == 'OPERATION ENABLED':
        return
    elif state == 'QUICK STOP ACTIVE':
        self.state = 'SWITCH ON DISABLED'
        self.state = 'OPERATION ENABLED'
    else:
        self.state = 'OPERATION ENABLED'

canopen.BaseNode402.enable_operation = enable_operation
```

af-silva, 2019-11-15:
Hi @gsulc nice to see you hacking the code,
This issue is related to #167, could you please check the code snippet I posted there and try it please, and validate?

Thanks!

gsulc, 2019-11-19:
This did not work. As it is a different case/transition. I've had limited time to try things out, but I'm working on something. I should have something at leas viable tomorrow.

gsulc, 2019-11-20:
Okay, here is the first pass that works:
```python
    __init__(blahblahbblah):
        #...
        self.use_state_transition_16 = True
        #...

    @state.setter
    def state(self, new_state):
        """ Defines the state for the DS402 state machine
        :param string new_state: Target state
        :param int timeout:
        :raise RuntimeError: Occurs when the time defined to change the state is reached
        :raise TypeError: Occurs when trying to execute a ilegal transition in the sate machine
        """
        t_to_new_state = time.time() + 8  # 800 milliseconds tiemout
        while self.state != new_state:
            try:
                if new_state == 'OPERATION ENABLED':
                    if not self.use_state_transition_16 and self.state == 'QUICK STOP ACTIVE':
                        next_state = 'SWITCH ON DISABLED'
                    else:
                        next_state = State402.next_state_for_enabling(self.state)
                else:
                    next_state = new_state
                
                self.controlword = State402.TRANSITIONTABLE[
                        (self.state, next_state)]
                
                # timeout of 400 milliseconds to try set the next state
                t_to_next_state = time.time() + 0.4
                while self.state != next_state:
                    if time.time() > t_to_next_state:
                        if new_state == 'OPERATION ENABLED' and self.use_state_transition_16:
                            self.use_state_transition_16 = False
                            continue
                        break
                    time.sleep(0.01)  # 10 milliseconds of sleep
            except KeyError:
                raise ValueError('Illegal transition from {f} to {t}'.format(f=self.state, t=new_state))
            # check the timeout
            except Exception as e:
                print(str(e))
            if time.time() > t_to_new_state:
                raise RuntimeError('Timeout when trying to change state')
            time.sleep(0.01)  # 10 miliseconds of sleep
```

There is an issue with this approach: if transition 16 fails for a different reason, but is a valid transition on the drive, the code wouldn't try it again. But having it always fail first and only using this as a fallback is also bad for performance. Having the API consumer specify if this one transition is available at construction time also seems obnoxious. This can also lead into the temptation to specify many other things this way, leading to a mess. Also, lets face it: most people aren't going to know.

My current thought is to add additional checking. If both transition 16 and 12 fail, then don't change anything. But if 16 fails and 12 succeeds, then stop using 16.

acolomb, 2021-09-08:
Just noticed this is related to my approach in #264, where I tried to generalize the state transitions.

Regarding this issue: I think such incomplete implementations could be handled on the application level, when the device is known to not support the direct transition.  Going to SWITCH ON DISABLED first, then back to OPERATION ENABLED  is just one line of code for the library API consumer (assuming #264 gets merged).  Trying to automate the solution here feels like bending over backwards just for a very small number of use cases.  Maybe better to add a hint to the relevant documentation that the manual transition might be needed first?

friederschueler, 2024-06-14:
Closing this as #264 is now merged and this issue is inactive.

closed 2024-06-14 by friederschueler