canopen-python/canopen #174: RxPDO sent value is updated slowly

Matiu-f, 2020-01-19:
Hello and thanks a lot for your work!

I've successfully configured a RxPDO to be sent every 10ms. The sent value should be different at any transmission, but I see on the slave node a received value that change every 3 or 4 sent PDOs (30-40ms).
Sending an SDO instead of a PDO I was able to send a different value every 1ms. (You can see the related code commented below.)

I leaved my previous tries as commented code.

Am I missing something?

Thanks in Advance!

`

	s60.rpdo.read()
	s60.rpdo[1].clear()
	s60.rpdo[1].add_variable('Target position')
	s60.rpdo[1].enabled = True
	s60.rpdo.save()
	s60.rpdo[1]['Target position'].raw = (7*6000).to_bytes(4, 'little')
	s60.rpdo[1].start(0.01)

	s60.nmt.state = 'OPERATIONAL'
	print("Gone Operational")

	network.sync.start(0.001)

	starttime = time.time()
	oldtime = time.time()
	while True:
		newtime = time.time()
		if newtime-oldtime>0.0005:
			cosminus1 = math.cos(1.0*(newtime-starttime))-1
			feeds = int(6000 * cosminus1 * 3.3 + 42000)
		#print(feeds, '@ timesector:', newtime-oldtime)
		#s60.sdo.download(0x607A,0,feeds.to_bytes(4, 'little'))		#MF: Target position

			#s60.rpdo[1].setdata = feeds.to_bytes(4, 'little')
			#s60.object_dictionary['Target position'].raw = feeds.to_bytes(4, 'little')
			#s60.rpdo[1].update()
			#s60.rpdo[1]['Target position'].raw = feeds.to_bytes(4, 'little')
			s60.rpdo[1]['Target position'].set_data(feeds.to_bytes(4, 'little'))
			oldtime = newtime
`

christiansandberg, 2020-01-19:
What type of CAN interface are you using?

Matiu-f, 2020-01-19:
Ixxat USB-to-CAN v2.

af-silva, 2020-01-20:
Hi @Matiu-f ,

What I normaly do is 

```python 

def target_position(self, value):
        """Target Position Setter used to set the target position of the node
        :param value: Position in counts (device internal units)
        """
        try:
            self.rpdo_pointers[0x607A].raw = value
            self.rpdo_pointers[0x607A].pdo_parent.transmit()
            self.controlword = State402.CW_OPERATION_ENABLED | self.CW_NEW_SETPOINT_CHANGE_IMMEDIATELY
            timeout = time.time() + 0.5
            while (self.statusword & self.SW_SETPOINT_ACKNOWLEDGE) != self.SW_SETPOINT_ACKNOWLEDGE:
                time.sleep(0.0001)
                if time.time() > timeout:
                    raise RuntimeError('Unable to set new position')
            self.controlword = State402.CW_OPERATION_ENABLED | self.CW_CHANGE_IMMEDIATELY
        except KeyError as ex:
            raise ValueError('The object 0x607A (Target Position) is not configured in any PDO, please check the device configuration! Error[{0}]'.format(ex.__str__()))
```

I think you are missing the ```s60.rpdo[1]['Target position'].transmit()```

Matiu-f, 2020-01-20:
Hello @af-silva ,
aren't you sending a single pdo in this way?
My goal is to enable the periodic automatic send (apparently it is already working correctly), but I find that the sent value is updated only every 30-40ms even if I update the Dictionary Object value "Target Position" every 1ms.

af-silva, 2020-01-20:
@Matiu-f 

Yes, but depending on your configuration with will only be executed at sync message. 
A think the way he framework  it's build, at sync it will grab the data and transmit it, so you should be seeing data being sent and updated at the frequency you configured at `network.sync.start()`

You have also to check the CAN interface configuration and the Bus speed supported, for instance I use SocketCAN  at 1000 Kbit/s ...
 
```python

s60.rpdo.read()
s60.rpdo[1].clear()
s60.rpdo[1].add_variable('Target position')
s60.rpdo[1].enabled = True
s60.rpdo.save()

s60.rpdo[1]['Target position'].raw = (7*6000).to_bytes(4, 'little')
s60.rpdo[1].start(0.01)

network.sync.start(0.01) # 10 milliseconds 

s60.nmt.state = 'OPERATIONAL'
print("Gone Operational")

starttime = time.time()
oldtime = time.time()

while True:
	delta = int(starttime  * 1000) - int(oldtime * 1000)
        if delta < 10:
	    continue
	cosminus1 = math.cos(1.0*(newtime-starttime))-1
	feeds = int(6000 * cosminus1 * 3.3 + 42000)
	s60.rpdo[1]['Target position'].set_data(feeds.to_bytes(4, 'little'))
	oldtime = newtime
```

christiansandberg, 2020-01-20:
Ixxat utilizes the hardware for periodic transmission which improves timing between messages, but unfortunately it does not provide a way to update the data after it has started. Therefore the message needs to be stopped and recreated every time the data is changed. This is probably very expensive and probably why you have such problems.

I don’t have a solution for you but implementing the periodic transmission yourself might be one.

Matiu-f, 2020-01-21:
Thanks Christian,
I can have access to a Peak PCan to USB too, do you think this interface could perform better?

christiansandberg, 2020-01-21:
It could be interesting to see if it behaves differently.

Matiu-f, 2020-01-22:
It seems it behaves worse: PCAN has a variable delay that reaches 70ms.
I took some time measurements:

PCAN SDO:
> data: 12097     deltatimems: 11
data: 41539     deltatimems: 13
data: 41588     deltatimems: 12
data: 41635     deltatimems: 12
data: 41699     deltatimems: 18
data: 41749     deltatimems: 15
data: 41796     deltatimems: 16
data: 41834     deltatimems: 14
data: 41863     deltatimems: 12
data: 41889     deltatimems: 12
data: 41947     deltatimems: 33
data: 41963     deltatimems: 12
data: 41999     deltatimems: 67
data: 41973     deltatimems: 45
data: 41955     deltatimems: 15
data: 41936     deltatimems: 13
data: 41906     deltatimems: 17
data: 41871     deltatimems: 17
data: 41698     deltatimems: 61
data: 41578     deltatimems: 32
data: 41269     deltatimems: 66
data: 41041     deltatimems: 40
data: 40948     deltatimems: 15
data: 40804     deltatimems: 22
data: 40629     deltatimems: 25
data: 40318     deltatimems: 40
data: 40213     deltatimems: 13
data: 40045     deltatimems: 20
data: 39941     deltatimems: 13
data: 39364     deltatimems: 61
data: 39184     deltatimems: 18
data: 38743     deltatimems: 42
data: 38308     deltatimems: 39
data: 38181     deltatimems: 11
data: 37714     deltatimems: 39

IXXAT PDO (today it behaves far better):
> data: 15440     deltatimems: 2
data: 15403     deltatimems: 2
data: 15366     deltatimems: 3
data: 15311     deltatimems: 3
data: 15255     deltatimems: 3
data: 15200     deltatimems: 2
data: 15163     deltatimems: 2
data: 15126     deltatimems: 3
data: 15071     deltatimems: 3
data: 15015     deltatimems: 3
data: 14960     deltatimems: 3
data: 14905     deltatimems: 3
data: 14850     deltatimems: 2
data: 14813     deltatimems: 2
data: 14777     deltatimems: 2
data: 14740     deltatimems: 3
data: 14685     deltatimems: 3
data: 14630     deltatimems: 3
data: 14576     deltatimems: 3
data: 14521     deltatimems: 3
data: 14467     deltatimems: 3
data: 14412     deltatimems: 2
data: 14376     deltatimems: 2
data: 14340     deltatimems: 2
data: 14303     deltatimems: 3
data: 14249     deltatimems: 2

Both the interfaces can send SDOs under the ms.

closed 2020-07-24 by af-silva