canopen-python/canopen #175: Lss Protocol: Identify slave

BenjaFischer, 2020-01-20:
https://github.com/christiansandberg/canopen/blob/61bc0a0874002963233d5401edf9c25fcdcea445/canopen/lss.py#L294

I was just testing my embedded lss protocol implementation with python.
In general it works fine, but I'm confused at one point. 
The fastscan acceps only "CS_IDENTIFY_SLAVE = 0x4F" as response from the slave.
"CS_IDENTIFY_NON_CONFIGURED_SLAVE = 0x50" is ignored although the fastscan is designed to find all non configured slaves.

As I could't find a clear descriptions when to use which command specifier, please check if the implementation is correct and may respond with your protocol description source.

Thanks in advance!

christiansandberg, 2020-01-20:
I have no knowledge in LSS, maybe @ymkim92 can help since he is the author.

ymkim92, 2020-09-09:
Slave in CS_IDENTIFY_NON_CONFIGURED_SLAVE state participate in fastscan.
while running fastscan, only one slave with the smallest number of identifier will respond with CS_IDENTIFY_SLAVE state.
I hope this will help you at not too late time.

friederschueler, 2024-06-14:
Closing this, as it is answered without further response.

closed 2024-06-14 by friederschueler