canopen-python/canopen #181: Getting same TPDO variable values on two different nodes

MillerEsp, 2020-03-31:
Hello,

I have problems when using PDOs with two nodes and a master. First of all, I configure each PDO of each node with the variables that I want to assign them, in the code you will see that I only configure, for each of the nodes, the first 3 TPDOs (currently, there are only 3 for the driver I am using) and then I leave the RPDOs disabled. In the first TPDO I have the variables 'actual_position' and 'position_demand' configured.

After the configuration I print the  ‘actual_position’ and ‘demand_position’ values ​​received by PDO, but in both nodes it prints the same value, and it is always the last one. That is, if I move motor1 , I get the 'actual_position' value from the motor1 on both TPDO1 (from node_1 and node_2) , and if I move the motor2, I get the 'actual_position' value from the motor2 on both TPDO1 (from node_1 and node_2). The same happens to me with the other variables.

Is it possible to receive the value of each variable in the TPDO of the corresponding node?

I'm using canopen v.1.0.0.

Any help is appreciated.

Thank you

af-silva, 2020-04-01:
Hi @MillerEsp,

More likely you have both motors configured with the same ID.
Each motor has an unique ID used by the CAN network to distinguish the messages originating and sent to the devices. If you have both motors with the same ID for a given TPDO you will see the same messages for both motors.
If you are loading the EDS file, make sure it does not defined the Node ID and make sure you have a correct configuration loaded in both motors (not the default one, is expected that the default configuration loads the same ID).

[image]
(Courtesy of [CAN-Cia](https://www.can-cia.org/can-knowledge/canopen-fd/pdo-protocol/) )

MillerEsp, 2020-04-02:
Hi af-silva, thank you for your answer.

I have been searching if the ID of the drive was defined in the EDS file, but I have not found its definition anywhere. Here is the link where the EDS file is located (from Ingenia's EVEREST XCR drive):
https://distext.ingeniamc.com/products/EVE-XCR/firmware/1.5.0/

On the other hand, when I start the script I check the list of available nodes, which prints the ids 10 and 11. This way I make sure they are detected, and then I add the node indicating the ID of each node and the EDS path .

Finally, note that all operations with SDOs work correctly for me, I send and receive data to the indicated node at any time without any problem.

Thank you very much again for your help.

*I've added the code snippet from the first comment. I had forgotten to add it.
     

        system("sudo ifconfig can0 txqueuelen 1000")
        self.__net = Network()
        system('sudo ip link set can0 type can bitrate 500000')
        system('sudo ifconfig can0 up')
        self.__net.connect(bustype='socketcan', channel='can0')

        # Add nodes
        self.__net.scanner.search()
        sleep(0.1)  # Wait a little while to allow node to respond
        node_list = self.__net.scanner.nodes
        print("Node list = " + str(node_list)) # print: [10, 11]

        self.node_1 = self.__net.add_node(self.node_1_id, self.__eds_path)
        self.node_2 = self.__net.add_node(self.node_2_id, self.__eds_path)
```

af-silva, 2020-04-02:
Hi @MillerEsp , no problem.

Ok, lets breakdown the problem.

```
    # Add nodes
    self.__net.scanner.search()
    sleep(0.1)  # Wait a little while to allow node to respond
    node_list = self.__net.scanner.nodes
    print("Node list = " + str(node_list)) # print: [10, 11]

    self.node_1 = self.__net.add_node(self.node_1_id, self.__eds_path)
    self.node_2 = self.__net.add_node(self.node_2_id, self.__eds_path)
```

First, I believe that `node_1_id` and `node_2_id` are 10 and 11 respectively.
Second, can you please try the DS402 profile to setup the nodes?? it should work without the DS402 but is just to try to understand what is happening, you have an example in the example folder.

Another thing you can try is:

```
   # Add nodes
    self.__net.scanner.search()
    sleep(0.1)  # Wait a little while to allow node to respond
    node_list = self.__net.scanner.nodes
    print("Node list = " + str(node_list)) # print: [10, 11]

    self.node_1 = canopen.RemoteNode(10, '/home/Code/test/your_eds_file.eds')
    self.__net.add_node(self.node_1_id, self.__eds_path)
    self.node_1.nmt.wait_for_bootup(15)

    self.node_2 = canopen.RemoteNode(11, '/home/Code/test/your_eds_file.eds')
    self.__net.add_node(self.node_2_id, self.__eds_path)
    self.node_2.nmt.wait_for_bootup(15)

    print('node state 10) = {0}'.format(self.node_2.nmt.state))
    print('node state 11) = {0}'.format(self.node_2.nmt.state))

```

Last, can you please sniff the messages going through the CAN Bus network, you can use the candump to see the messages going to and from the master to the remote nodes. 

The sample codes always worked for me, I'm also using Ingeniamc controllers, in my case Jupiter.
For to help a bit more please turn on the Debug flag, and sinff the CAN Bus to see what messages are being sent and from where.

(Sorry for not being much of a help right now, but need a bit more info)
Thanks ;)

MillerEsp, 2020-04-14:
Hi @af-silva.

Sorry for my late answer.

First of all, thank you very much for all your suggestions and your help :).

Second, I have tested the code you have inside the second box and I get the following error:

```
 in __configure_pdos_sync
    self.node1.tpdo.read()
  File "/usr/local/lib/python3.7/dist-packages/canopen/pdo/base.py", line 55, in read
    pdo_map.read()
  File "/usr/local/lib/python3.7/dist-packages/canopen/pdo/base.py", line 288, in read
    cob_id = self.com_record[1].raw
  File "/usr/local/lib/python3.7/dist-packages/canopen/variable.py", line 74, in raw
    value = self.od.decode_raw(self.data)
  File "/usr/local/lib/python3.7/dist-packages/canopen/variable.py", line 36, in data
    return self.get_data()
  File "/usr/local/lib/python3.7/dist-packages/canopen/sdo/base.py", line 108, in get_data
    return self.sdo_node.upload(self.od.index, self.od.subindex)
  File "/usr/local/lib/python3.7/dist-packages/canopen/sdo/client.py", line 117, in upload
    fp = self.open(index, subindex, buffering=0)
  File "/usr/local/lib/python3.7/dist-packages/canopen/sdo/client.py", line 197, in open
    raw_stream = ReadableStream(self, index, subindex)
  File "/usr/local/lib/python3.7/dist-packages/canopen/sdo/client.py", line 243, in __init__
    response = sdo_client.request_response(request)
  File "/usr/local/lib/python3.7/dist-packages/canopen/sdo/client.py", line 82, in request_response
    self.send_request(sdo_request)
  File "/usr/local/lib/python3.7/dist-packages/canopen/sdo/client.py", line 53, in send_request
    self.network.send_message(self.rx_cobid, request)
AttributeError: 'NoneType' object has no attribute 'send_message'

```

On the other hand, I have sniffed the message with KVASER and KVASER CanKING and it seems that I get the same COBID when I save the configuration of both TPDOs. Maybe is this the problem?

TPDO Node 1:

Chn | Identifier | Flg | DLC | D0 | D1 | D2 | D3 | D4 | D5 | D6 | D7 | Time | Dir
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
0 | 0 |   | 2 | 80 | 0A |   |   |   |   |   |   | 2574.01105 | R
0 | 0000060A |   | 8 | 23 | 00 | 18 | 01 | **A0** | **01** | **00** | **80** | 2574.0119 | R
0 | 0000058A |   | 8 | 60 | 00 | 18 | 01 | 00 | 00 | 00 | 00 | 2574.01215 | R
0 | 0000060A |   | 8 | 2F | 00 | 18 | 02 | 00 | 00 | 00 | 00 | 2574.01463 | R
0 | 0000058A |   | 8 | 60 | 00 | 18 | 02 | 00 | 00 | 00 | 00 | 2574.01489 | R
0 | 0000060A |   | 8 | 2B | 00 | 18 | 03 | E8 | 03 | 00 | 00 | 2574.0171 | R
0 | 0000058A |   | 8 | 60 | 00 | 18 | 03 | 00 | 00 | 00 | 00 | 2574.01736 | R
0 | 0000060A |   | 8 | 2B | 00 | 18 | 05 | 00 | 00 | 00 | 00 | 2574.01949 | R
0 | 0000058A |   | 8 | 60 | 00 | 18 | 05 | 00 | 00 | 00 | 00 | 2574.01975 | R

TPDO Node 2

Chn | Identifier | Flg | DLC | D0 | D1 | D2 | D3 | D4 | D5 | D6 | D7 | Time | Dir
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
0 | 0 |   | 2 | 80 | 0B |   |   |   |   |   |   | 2602.60369 | R
0 | 0000060B |   | 8 | 23 | 00 | 18 | 01 | **A0** | **01**| **00** | **80** | 2602.60454 | R
0 | 0000058B |   | 8 | 60 | 00 | 18 | 01 | 00 | 00 | 00 | 00 | 2602.60479 | R
0 | 0000060B |   | 8 | 2F | 00 | 18 | 02 | 00 | 00 | 00 | 00 | 2602.60718 | R
0 | 0000058B |   | 8 | 60 | 00 | 18 | 02 | 00 | 00 | 00 | 00 | 2602.60744 | R
0 | 0000060B |   | 8 | 2B | 00 | 18 | 03 | E8 | 03 | 00 | 00 | 2602.60957 | R
0 | 0000058B |   | 8 | 60 | 00 | 18 | 03 | 00 | 00 | 00 | 00 | 2602.60983 | R
0 | 0000060B |   | 8 | 2B | 00 | 18 | 05 | 00 | 00 | 00 | 00 | 2602.61196 | R
0 | 0000058B |   | 8 | 60 | 00 | 18 | 05 | 00 | 00 | 00 | 00 | 2602.61222 | R

Do you know where the problem is? As I said in previous comments, I use different node IDs for each node and the SDO messages work without problems. I don’t know why the COBID is the same for both.

Thank you very much again!

af-silva, 2020-04-14:
Hi @MillerEsp ,

> Second, I have tested the code you have inside the second box and I get the following error:

`AttributeError: 'NoneType' object has no attribute 'send_message'`
The  `self.__net` is not set I think, from the error message, I just post a snipped, not the all code.

```
    # Setup network
    system("sudo ifconfig can0 txqueuelen 1000")
    self.__net = Network()

    system('sudo ip link set can0 type can bitrate 500000')
    system('sudo ifconfig can0 up')
    self.__net.connect(bustype='socketcan', channel='can0', bitrate = 500000)

    self.network.check()

    # Add nodes
    self.__net.scanner.search()
    sleep(0.1)  # Wait a little while to allow node to respond
    
    # Check if nodes are configured
    node_list = self.__net.scanner.nodes
    print("Node list = " + str(node_list)) # print: [10, 11]

    self.node_1 = canopen.RemoteNode(0x10, '/home/Code/test/your_eds_file.eds')
    self.__net.add_node(self.node_1_id, self.__eds_path)
    self.node_1.nmt.wait_for_bootup(15)

    self.node_2 = canopen.RemoteNode(0x11, '/home/Code/test/your_eds_file.eds')
    self.__net.add_node(self.node_2_id, self.__eds_path)
    self.node_2.nmt.wait_for_bootup(15)

    print('node state 10) = {0}'.format(self.node_2.nmt.state))
    print('node state 11) = {0}'.format(self.node_2.nmt.state))

    self.network.sync.start(0.01)

```

Ok, seems the problem is indeed with the Node ID's, in this case the COB-ID generation... I'm trying to understand why is this happening. In the meantime can you try to use the code above or incorporate the changes, the `wait_for_bootup` in your code ?

Some troubleshooting, the eds file has the right permissions to be read?
Can you try to change the eds file content to specify the correct NODEIDvalue?

Search for **NODEID** and add `ParameterValue= <value of your node>` in this case **A** (10) and **B** (11)
```
[1802sub1]
ParameterName=COB-ID used
ObjectType=0x7
DataType=0x0007
AccessType=rw
DefaultValue=$NODEID+0x40000380
PDOMapping=0x0
ParameterValue=0x4000038A
```
Note: create two eds files, one for each motor
 

> Do you know where the problem is? As I said in previous comments, I use different node IDs for each node and the SDO messages work without problems. I don’t know why the COBID is the same for both

I'm also confused... i'm missing something, now that i look at your initial code, when you first read the TPDO and RPDO 
```
    self.node_1.tpdo.read()
    self.node_1.rpdo.read()
    self.node_2.tpdo.read()
    self.node_2.rpdo.read()
```
This uses the SDO server to get the configured nodes from the Device, and you were doing it correctly. I had problems in the past with Ingeniamc firmwares, I like the controllers but some times they also make mistakes, they are humans after all. So try to force the NODEID in the EDS file and load the file, or you can set the eds file to null when creating the nodes `canopen.RemoteNode(10, null')`, this will create a empty object dictionary, you have to configure the node like you did doh. 

Try this things and let me now, I never experience this, i had other problems like some Object in the dictionary not holding a value like it was supposed to, but since they fixed that i'm using this library, with my implementation without any problems.

Good hacking ;)

MillerEsp, 2020-05-05:
Hi @af-silva,

Sorry to reply so late again.

First of all, thank you very much for your answer, we have finally managed to change the COBID manually using SDOs following the instructions on the Ingenia page, and now it works :). Thank you very much for your help, this problem has finally been solved.

On the other hand, I wanted to know if it is possible to use PDOs without using an EDS file, but directly, using just the registers. I have tried but the **clear()** function stops working.

Thank you very much again!

af-silva, 2020-05-05:
Hi @MillerEsp 

If you set the EDS to null there is nothing load into the Maps and so nothing to **Clear**.

This what the clear function does, just set the map to an empty array.
```
    def clear(self):
        """Clear all variables from this map."""
        self.map = []
        self.length = 0
```

Try to read the PDO's configuration loaded in the drives using the `import_from_node` function by providing the _node_id_ and _network_ objects. The functions uses SDO's calls to the remote note do get the PDO's configuration and stores it in the the object dictionary.

friederschueler, 2024-06-13:
Problem seems to be solved.

closed 2024-06-13 by friederschueler