canopen-python/canopen #183: Reading op mode in CiA 402 profile causes KeyError when not set to a specific mode

allesblinkt, 2020-04-24:
When the current operation mode of a device using the 402 profile is read using `mode = thing.op_mode` it might be in a mode which is not mapped by `class OperationMode`. Therefor a `KeyError` might be thrown. Most prominently this happens to me when a motor has been without power and has no mode (=0) set yet.

```python
Traceback (most recent call last):
  File "/home/heiner/heiner-control/heiner_control/canopen/master.py", line 33, in init
    slave.init()
  File "/home/heiner/heiner-control/heiner_control/canopen/axis.py", line 89, in init
    node.op_mode = 'PROFILED POSITION'
  File "/home/heiner/heiner-control/env/lib/python3.7/site-packages/canopen/profiles/p402.py", line 339, in op_mode
    while self.op_mode != mode:
  File "/home/heiner/heiner-control/env/lib/python3.7/site-packages/canopen/profiles/p402.py", line 301, in op_mode
    return OperationMode.CODE2NAME[self.sdo[0x6061].raw]
KeyError: 0
```
To mitigate a part of this problem, I have added `NO MODE` as a valid mode in in the `OperationMode` class.

af-silva, 2020-04-27:
Hi @allesblinkt ,

The motor should always have the mode set internally and/or loaded by the eds file. But I understand the behavior you described and i think it can depend on the way manufacturers implement their drive. That being said, I think this has no secondary effects with the normal procedures to set up a driver and serves as a default behavior to avoid errors.

Thanks for identifying this, i will accept your PR.

closed 2020-04-27 by af-silva

allesblinkt, 2020-04-28:
Hello @af-silva. Thank you!