canopen-python/canopen #188: Socketcanopen vs Canopen

amostafa9, 2020-05-20:
What is the difference between socketcanopen and canopen? Are they interchangeable? 

In the read the docs i see import canopen but in the examples from the github i see import socketcanopen.

af-silva, 2020-05-21:
Hi @amostafa9,

What examples ???
CANOpen is a communication protocol, and i think you want to say SocketCAN, that is a "generic" CAN driver for Linux that provides the communication interface. You can configure the CANOpen to use different drivers such as socketcan, kvaser and others (see this [link](https://python-can.readthedocs.io/en/stable/))

From readthedoc:
``` python
network.connect(channel='can0', bustype='socketcan')
# network.connect(bustype='kvaser', channel=0, bitrate=250000)
# network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate=250000)
# network.connect(bustype='ixxat', channel=0, bitrate=250000)
# network.connect(bustype='nican', channel='CAN0', bitrate=250000)
```  
The CANOpen API uses the SocketCAN driver as a interface through **python-can** to communicate with the CAN bus network.

amostafa9, 2020-05-21:
Thanks for the response. I was thrown off a bit by this repo: https://github.com/bggardner/canopen-rpi
Looks like this also uses socketcan but specifically for the RPI. I guess it still uses the CANopen for python API as its library for CANopen?

The examples i was referring to were on that github page.

af-silva, 2020-05-21:
I don't know that API ...
But like a said, CANOpen is a protocol, SocketCAN is a "generic" driver for Linux that enables the communication with the CAN bus network.

From the readme file from the [canopen-rpi](https://github.com/bggardner/canopen-rpi)

> socketcan, abstracts the CAN interface by providing Bus and Message classes. The socketcan module can be used to create adaptors to translate CAN traffic to other protocols. The second module, socketcanopen, contains classes to instantiate a CANopen node application. Each node must be initialized with at least one socketcan.Bus, a node ID (integer, 1 to 127), and an socketcanopen.ObjectDictionary.

Is exactly what i said, **socketcan** for access the CAN bus network and **socketcanopen** implements the CANOpen protocol, the name perhaps is confusing ...

This API (i.e., the canopen API) provides the implementation for interaction with CANOpen enabled devices, supporting the CiA 301 and DS-402 profile.

amostafa9, 2020-05-21:
Thank you. Another question. Is there slave support with this API?
How can I instantiate a slave? 
If there isn't slave support, how can I at least implement slave functionality with this API? Thanks for the help.

af-silva, 2020-05-21:
No problem. 

Slave nodes are supported bu limited, you can instantiate a slave node by using the LocalNode class. 
For more information regarding the state of the implementation check #47 .
At this point, this library is meant be used as a master controlling slave nodes, but you can starting tinkering with the code and help improving the support for slave nodes locally. 
Good hacking ;)

bggardner, 2020-06-01:
The `socketcanopen` Python module (which I developed independently from this repo), is targeted for creating both masters (including flying masters) and slaves using the same `Node` class.  Most of the node operation is derived from the object dictionary, so it is somewhat "automated".  I just changed the is to use the `python-can` module instead of my "in-house" `socketcan` module.  I have some examples of slaves and masters.  However, testing has been limited to using socketcan drivers.

I would love to merge the features of the two repos into one (preferably this one), since it's already registered in PyPI.  I'll post an update here if/when I create a fork.

friederschueler, 2024-06-14:
Closing this for now, we still have the slave node support issue #47

closed 2024-06-14 by friederschueler