canopen-python/canopen #189: SDO Write

amostafa9, 2020-05-28:
I'm trying to use SDO to send the value "12" from the Master to the Slave. 

I'm using this command:
node.sdo[0x6401][1].write(12,'phys')

The command executes fine, and it looks like the slave registers the SDO request because i see the Expedited Download in the canopen logger.

However, on the slave side I have a continuous print statement in a while loop that prints the value of the OD at [0x6401][1] and it never changes to 12.  The print statement on the slave side is:
print(local_node.object_dictionary[0x6401][1].value)

My slave node is not in OPERATIONAL mode, because I didn't think that for SDOs to work the slave had to be operational.

Why is that print statement never changing to 12? Any idea what I'm doing wrong?
What is the best way to write a value to a slave node?

christiansandberg, 2020-05-28:
The new values are not currently stored in the Object Dictionary, its values are whatever the EDS file specified. You can access the current value on the local node using the .sdo object just like for the remote node. The idea is to have the same API for local and remote nodes but it could make sense to connect it to the Object Dictionary.

amostafa9, 2020-05-28:
Thanks! I was able to do local_node.sdo[0x6401][1].phys and obtain the value from the master.

One could always store the variable manually in the local OD once it has been received.

closed 2020-05-28 by amostafa9