canopen-python/canopen #193: Network.connect() overwrites bus supplied with Network constructor

bggardner, 2020-06-03:
[This line](https://github.com/christiansandberg/canopen/blob/master/canopen/network.py#L109) in `Network.connect()` overwrites the "re-used" bus that may have been provided in the `Network` constructor.  Therefore, if just `connect()` is called with no arguments (as shown first in the README), a `NotImplementedError: Invalid CAN Bus Type - None` is raised.  `connect()` should either check if a bus is already set, or if there are arguments passed before overwriting the `bus` property.

christiansandberg, 2020-06-03:
The idea was that you should either provide your own bus and Notifier _or_ use the connect method, but not both. Do you think there is a use case for using both? Or could the documentation be clearer?

bggardner, 2020-06-03:
Maybe the documentation needs to be clearer, or at least remove the uncommented `network.connect()` example from the README, since it really requires arguments to instantiate a bus, unless one of my recommendations were implemented.

In my case, I created a custom `can.BusABC` class, and passed the instantiation to `Network()`.  To be able to use it with `canopen.sdo.SdoClient`, I had to then manually set `network.notifier`, which is basically what `Network.connect()` does.

friederschueler, 2024-06-14:
Fixed with #404

closed 2024-06-14 by friederschueler