canopen-python/canopen #194: High school student asking for help

didigithub, 2020-06-13:
I am a high school student and am making a robotic arm. I use python-canopen to control motors in CiA 402 CANopen device. I read the document carefully, but I am still confused. Can you send me a complete example code to help me understand python-canopen better? Thank youverymuch。                 
18616707393@163.com

af-silva, 2020-06-15:
Hi @didigithub, first and foremost you need to understand how the CiA 402 protocol works, in particular the state machines that are involved in managing the communication and device states. Second, you can look at this thread #167 for the state transitions, and little search for DS402 State transitions and DS402 object dictionary will be enough to get you started. Also, if you are communicating with slave nodes you need to read the documentation for the that particular device to check which configurations they use (this is, which  objects are mapped). 

For a code example, you have a snippet in the **_examples folder_**, and is a working example, for you to have more control over the nodes execution I strongly suggest you to extend the **BaseNode402** and add the functionality you need for specific types of control, for instance control nodes using position or velocity.

Hope this helps

af-silva, 2020-07-24:
Closing this issue, if you still need help reopen.

closed 2020-07-24 by af-silva