canopen-python/canopen #195: question

mkeyno, 2020-06-23:
Hi
sorry to open a ticket for my question, I've been trying to use your lib to communicate with my CAN device with my [USB-CAN adaptor](https://www.ifm.com/nz/en/product/EC2112?tab=documents) , actually its make COM port of my control panel, so I'm thinking I should open serial connection then configure CAN network parameters 
can you advise me how to set up it and scan and read CAN device with this medium 

BR

af-silva, 2020-06-30:
Hi @mkeyno ,

Take a look [here ](https://python-can.readthedocs.io/en/stable/interfaces/serial.html) or [here](https://python-can.readthedocs.io/en/stable/interfaces/usb2can.html)

I don't have experience with that device in particular, but you need the correct driver (linux or windows) to access the CAN network. This library uses the Python-Can as an interface that support many different ways to access the CAN network. Read the documentation provided and follow the steps that best fit your scenario.

af-silva, 2020-07-24:
Closing this issue, if you still need help please reopen.

closed 2020-07-24 by af-silva