canopen-python/canopen #196: how to confogure Io control?

Jiongyu, 2020-07-28:
**1. code:**
` self.__node.sdo[0x2193][0x01].raw = 0x02`
`self.__node.sdo[0x2193][0x02].raw = 0x02 `

**2. error:**
Traceback (most recent call last):
  File "/home/tan/ros/gripper_control/src/gripper_control/canopen_communication/scripts/test_script.py", line 31, in <module>
    main()
  File "/home/tan/ros/gripper_control/src/gripper_control/canopen_communication/scripts/test_script.py", line 13, in main
    T5.start()
  File "/home/tan/ros/gripper_control/src/gripper_control/canopen_communication/modular/modular_T85.py", line 60, in start
    self.__param_config()
  File "/home/tan/ros/gripper_control/src/gripper_control/canopen_communication/modular/modular_T85.py", line 235, in __param_config
    self.__node.sdo[0x2193][0x01].raw = 0x02 
  File "/usr/local/lib/python2.7/dist-packages/canopen/variable.py", line 88, in raw
    self.data = self.od.encode_raw(value)
  File "/usr/local/lib/python2.7/dist-packages/canopen/objectdictionary/__init__.py", line 350, in encode_raw
    value, self.data_type))
TypeError: Do not know how to encode 2 to data type Ah

**3. #The dictionary：**
[2193sub0]
ParameterName=Number of Entries
ObjectType=7
DataType=5
AccessType=RO
PDOMapping=0
DefaultValue=8
LowLimit=8
HighLimit=8

[2193sub1]
ParameterName=Output 0 configuration
ObjectType=7
DataType=10
AccessType=RW
PDOMapping=1

[2193sub2]
ParameterName=Output 1 configuration
ObjectType=7
DataType=10
AccessType=RW
PDOMapping=1

christiansandberg, 2020-07-28:
What do those objects do? They seem to be defined as OCTET_STRING types which means you need to write the data as a bytes object rather than an integer.

Is it possible that it is supposed to be 0x10 = UNSIGNED24?

Jiongyu, 2020-07-28:
those objects are used to set the output state configuration. when I use the form of can-message to sent those command, I got the following error:
**code:** 
```
        # self.__node.sdo.download(0x2193, 0x01, b'\x00\x02')
        # self.__node.sdo.download(0x2193, 0x02, b'\x00\x02')
```
**error:**
`        0x08000000: "General error"`
what is the mean of the general error?

christiansandberg, 2020-07-28:
The error comes from the node as an abort code so it's not very helpful. But it could be that you need to swap the bytes if it is little endian coded, I.e. 02 00.

Jiongyu, 2020-07-28:
when I change the code to little-endian, the general error still happen.  how do I find the reason which causes the error?

Jiongyu, 2020-07-28:
how can I sent the "OCTET_STRING" message? can you give me some code example?

christiansandberg, 2020-07-28:
I think the easiest way is to read the value first to see how it looks like and then write using the same format.

closed 2020-08-02 by Jiongyu