canopen-python/canopen #197: In SDO download command not possible send binary negative value

daddi1987, 2020-07-28:
Trying to use the SDO download function I can't send a binary negative value. I always answer this error:

- canopen.sdo.exceptions.SdoAbortedError: Code 0x06070012, Data type does not match, length of service parameter too high

My code works like this:
`MOTOR_LOADER.sdo.download(0x3001, 0, "b'\xff\xff\xff\xff\xff\xff\xff\x02'")`

[image]

christiansandberg, 2020-07-28:
It seems you need to remove the "" around the bytes.

daddi1987, 2020-07-29:
Sorry, the screen is referment of a test string. 
The correct string is:
`MOTOR_LOADER.sdo.download(0x3001, 0, b'\xff\xff\xff\xff\xff\xff\xff\x20')`

and error is
[image]

christiansandberg, 2020-07-29:
And you're sure the variable is a 64-bit integer? What if you try to upload the current value?

daddi1987, 2020-07-30:
sorry, the problem was with my driver set to 16 Bit. Congratulations on the excellent CanOpen library and support.

closed 2020-07-30 by christiansandberg