canopen-python/canopen #199: Read updated OD data from application

mhadavan, 2020-08-06:
Hello all!
First of all, thanks for the great work creating this library!!

I'm a novice regarding Canopen, so forgive me if my questions are stupid!

I have been playing around with this library for some days now, and I have a questions related to reading the SDO/OD data from the application:

In my current test I'm performing a expedited SDO download from my remote note to my PC(local node).
But I can't figure out how to fetch the data from my application.

Can log shows 4 bytes of data is downloaded to index 6555, subindex 1:
` vcan0  610   [8]  23 55 65 01 AA BB CC DD`
` vcan0  590   [8]  60 55 65 01 00 00 00 00`

I have tried to read using following code:
`self.node = self.cannetwork.add_local_node(nodeid, edsfile)`
`....code between...`
`data = self.node.sdo[0x6555][1].value`

But I always get "TypeError: 'NoneType' object is not subscriptable", even if I try data = self.node.sdo[0x6555][1].

I can see that the data is received as I get the following logging message:
`2020-08-06 07:28:17,323 : can.notifier- [server.py - init_download - lineNo 134]`
`INFO : Expedited download for 0x6555:1 `

Also If I add a write callback, I'm getting a call to my callback when I'm receiving the data, and the data is correct.

Parameter in EDS file:
`[6555]`
`ParameterName=Test`
`ObjectType=0x8`
`SubNumber=2`

`[6555sub0]`
`ParameterName=Test data 1`
`ObjectType=0x7`
`DataType=0x0007`
`AccessType=ro`
`DefaultValue=0x00000000`
`PDOMapping=0`

`[6555sub1]`
`ParameterName=Test data 2`
`ObjectType=0x7`
`DataType=0x0007`
`AccessType=rww`
`DefaultValue=0x00000000`
`PDOMapping=0`

af-silva, 2020-08-10:
Hi, 

Have you try the following:

1st: set the OD before trying to get the data?
2nd: download the OD before trying to get the data?

But first try to see what is in `node.sdo[0x6555].raw` ...
The correct way is what you have done in the first call -> `data = self.node.sdo[0x6555][1].value`, but i think something else is at play here. 

Try to use candump or other sniffer to see what is being exchange between your local node and remote.

af-silva, 2020-09-25:
No response/feedback provided.

closed 2020-09-25 by af-silva