canopen-python/canopen #201: RPDO Read fails when not all implemented

Rokmonkey, 2020-08-20:
I have several motors that have DCF files from the configuration software.  For whatever reason, neither the EDS or DCF files are completely filled out as the module demands.  For example, the DCF file has TPDO 22 configure and only that, so 0x1A00 is not in the DCF.  This throws an error when creating a BaseNode402 object or when trying to read RPDO/TPDO if the node is created with network.add_node().

The error given is as expected:
canopen.sdo.exceptions.SdoAbortedError: Code 0x06020000, Object does not exist

I believe this is because certain SDOs are hard-coded in the module for the TPDO/RPDOs instead of reading from the [OptionalObjects] section in the DCF/EDS.  I don't know the module well enough to make a pull request and I may also be incorrect in my assumption.  Any suggestions on addressing?

af-silva, 2020-08-24:
Hi @Rokmonkey ,

> It must not be from reading the EDS, as they don't show up anywhere.

It seems you are reading the configuration from the driver itself and not from the EDS DCF file, `self.pdo.read()  # reads the new configuration from the driver`.

And I don't think your problem is related to the TPDO's and RDPO's configuration .... 
I think you are having problem when uploading the configuration to the driver `return self.sdo_node.upload(self.od.index, self.od.subindex)`, so if you can please check what is happening when this is being fired, and see what is passing in the canbus network. 

Double check if you are tying to read the configuration from the driver? (if you do not pass the OD it will try to read from the remote note it's connecting to)

Rokmonkey, 2020-08-24:
Ahh, perhaps that's it.  However, that self.pdo.read() happens automatically when creating a node with BaseNode402, there isn't anything I can do with the library as is.

What is the proper way to read the PDO's from a DCF?

af-silva, 2020-08-25:
Hi @Rokmonkey,

The pdo.read() from the driver should only happen if you don't provide a ObjectDictionary, if you provide the OD then the code will try to send it to the driver and then read it to make sure the configurations have been set on the driver (configuration of the remote nodes). Try the below code, if this not resolve you issue try to sniff the canbus network to see what is being sent and what is causing the return of the Object does not exist error, namely which Object is returning this. 

```                    
vnode = VNode("10", "path_to_eds")
self.network.add_node(vnode)
vnode.nmt.state = 'RESET COMMUNICATION'  # Reset network
vnode .load_configuration
```
or, in the signature of the class instance constructor you have the flag to load the configuration (by default is false) `def __init__(self, node_id, object_dictionary, load_od=False):` so:
```
node = BaseNode402("10", "path_to_eds", true)
```

af-silva, 2020-09-25:
Hi @Rokmonkey, did you manage to solve the problem? Can I close this ticket?

friederschueler, 2024-06-14:
Closing for inactivity.

closed 2024-06-14 by friederschueler