canopen-python/canopen #202: Buffer overrun issue

mumeh996, 2020-09-21:
Using CANopen library with Socketcan on an RPi running Raspbian Buster with Kvaser Leaf Lite - connected to single axis of a CAN402 motion module (Copley STM).  It generally works well and is quite well documented.  

That said, I am having trouble figuring out why I am overrunning the internal buffer on the RPi.  I did not encounter this right away, but rather after several homes and moves (somewhere between 10-20).  I am using the basic code structure showcased in your documentation, and it works fine for a period of time, but then I receive **Errno 105 - no buffer space available**.  In the error log, I see RPDO commands being sent, so the source of the issue remains a mystery.

Also, the buffer is referenced in the documentation, but it is unclear if there is a way to clear said buffer.

Has anyone else run into this?

mumeh996, 2020-09-21:
I've done a little more analysis on my end and discovered that the motion drive/module is the offending party, in that it is offering verbose status chatter on the CANopen network (unsolicited, it appears).  I am attacking this on the drive side of things, but is there a way **(other than the brute force of using a network.disconnect after each transmission/query)** to enlarge or manage the inbound CANopen comms buffer?

af-silva, 2020-09-23:
Hi @mumeh996,

I have to check, but the network implementation uses the [https://github.com/hardbyte/python-can](python-can) library. I would suggest, if you have some time, to explore the python-can and check if it has some sort of mechanism to handle the comms buffer, and if so you can implement in the network a call to that mechanism...
Right now i don't have much time to look into it, but I will let the tick open and revisit when I can, or if in the meanwhile you have a solution I will close it.

Sorry for not be of much help. 
Regards.

friederschueler, 2024-06-13:
> Errno 105 - no buffer space available

This is an OS error, you can for example try to increase the buffer of the can device (can0) on a Linux system:

    ip link set can0 txqueuelen 1000

I will be closing this as there was no further reaction and this is a question.

closed 2024-06-13 by friederschueler