canopen-python/canopen #203: Prediotic Sync message is missing id

illuusio, 2020-10-13:
Is perdiotic Sync message correct as it only sends 0x80 and DLC is zero but spec expects it to be like sync.py trasmit added by one every transfer so DLC should be 1. If it doesn't have any data it breaks some CANBus controllers.

[1] https://www.can-cia.org/can-knowledge/canopen-fd/sync-protocol/

christiansandberg, 2020-10-16:
We don't support CANopen FD but if someone wants to investigate what needs to be done to make that happen that would be great! For regular CANopen it is optional:

> By default, the SYNC message does not carry any data (DLC = 0).  Devices that support CiA 301 Version 4.1 or higher may optionally offer a SYNC message, which provides a 1 byte SYNC counter value.

https://www.can-cia.org/can-knowledge/canopen/special-function-protocols/

illuusio, 2020-10-19:
Ok so this is allowed message and CANBus controller is not working correctly. I have to ask from CANBus controller manufacturer then why it produces error.

christiansandberg, 2020-10-19:
It's still a feature that could be implemented so I'm reopening it again. The issue is that right now we try to delegate the transmission to the kernel or hardware if possible for better timing. This means however that we can't update the data for each message easily. Then we would need to handle the periodic transmission in a Python thread instead which will make the sync message periodicity jitter a lot.

illuusio, 2020-10-20:
Should there that send_periodic-method would take also Callable (method) besides list as data parameter so there could be callback method in sync.py which feeds needed data per sending? I don't know if it's doable but it wouldn't need new thread.