canopen-python/canopen #205: sdo.phys or sdo.raw can't input negative value

tshh, 2020-10-19:
'max speed' take positive or negative value
positive value works ok
but, negative value failed to work

try:
node.sdo['max speed'].raw=-1

output:
Traceback (most recent call last):
  File "/home/tshh/miniconda3/envs/canopen/lib/python3.6/site-packages/canopen/objectdictionary/__init__.py", line 342, in encode_raw
    return self.STRUCT_TYPES[self.data_type].pack(value)
struct.error: argument out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tshh/miniconda3/envs/canopen/lib/python3.6/site-packages/canopen/variable.py", line 88, in raw
    self.data = self.od.encode_raw(value)
  File "/home/tshh/miniconda3/envs/canopen/lib/python3.6/site-packages/canopen/objectdictionary/__init__.py", line 344, in encode_raw
    raise ValueError("Value does not fit in specified type")
ValueError: Value does not fit in specified type

christiansandberg, 2020-10-19:
How is max speed defined in the EDS file? I suspect it is defined as an unsigned integer.

tshh, 2020-10-21:
indeed!

[6003]
ParameterName=max speed
ObjectType=7
DataType=0x0007
AccessType=rw
DefaultValue=0
PDOMapping=1

christiansandberg, 2020-10-21:
Try changing data type to 4 instead if you know that negative speeds really should be allowed.

closed 2020-10-24 by christiansandberg