canopen-python/canopen #207: Lokking for EDS examples - Jupiter.eds?

DaveTheCave, 2020-11-23:
In the example "canopen/examples/simple_ds402_node.py "

There is an eds file that I cannot find

'/home/andre/Code/test/jupiter.eds'

Am I just not looking in the right places?

Or is it missing?

I would appreciate a simple master slave network example with EDS file

CAn you give me any pointers?

Dave

christiansandberg, 2020-11-24:
Do you have an actual CANopen node that you want to test with? In that case you should request an EDS file from the manufacturer.

af-silva, 2020-11-24:
Hi @DaveTheCave, that is my fault. I will try by the end of today provide a simple example EDS file and correct the path to it in the simple_ds402_node.py. By Like Christian said, you should get an EDS file for your specific drive, since there are many objects in the OD that are specific for each manufacture. If you need one sonner, please google search one, there are many available from different manufacturers. 

Thanks.

af-silva, 2020-11-25:
EDS example file provided, and fixed paths to the new example file. Be aware that this file was modified and is not in any means a valid EDS file for drivers. If you are working with a device, please get the official EDS file from the manufacture and use it instead.

closed 2020-11-25 by af-silva