canopen-python/canopen #208: How to capture sent message

blairfancy, 2020-12-01:
Hi any way to capture sent message for file logging purpose? With can library it is easy has can message are clear but in canopen has we manipulate only object we do not have access to real can message sent.
What i do in can is to put all sent and received message to a queue object for further traitement.
Can we do the same in canopen?

hope my question is clear enough.

christiansandberg, 2020-12-01:
Which CAN interface are you using? Some of them support `receive_own_messages` and in combination with a listener you could be able to log all messages.

blairfancy, 2020-12-02:
I use Ixxat USB can interface. receive_own_messages = True seems not usable on this interface.

christiansandberg, 2020-12-02:
Then you should be able to add `receive_own_messages=True` to `network.connect()`, then add a callback function to `network.listeners`. Let me know if you need more guidance.

blairfancy, 2020-12-04:
Ok after several attempt i got SDO tx message in my log but unfortunately it does not work with NMT messages.

friederschueler, 2024-06-14:
Your own `MessageListener` will fetch all the raw can messages, so you should be able to receive nmt messages as well.

Closing this for inactivity. If your problem persists, please open a new issue with detailed description, code and output logs.

closed 2024-06-14 by friederschueler