canopen-python/canopen #209: Using PDO to send data from host PC to device.

SoumyaDeshapnde, 2020-12-07:
I'm using the PEAK PCAN USB adapter. I'm trying to use PDO to send messages to the board from my host PC.

Anu suggestions? I'm new to CAN as well as python.

blairfancy, 2020-12-08:
I had same issue. you need to transmit message after set it:
`nodes[2].rpdo[1].transmit()`

SoumyaDeshapnde, 2020-12-08:
So, creating 2 nodes: 1 host PC and 1 device?
Will the host PC be a local node? canopen.LocalNode(1, /path_to_eds) 
or just remote 2 nodes? canopen.RemoteNode(2, /path_to_eds)

blairfancy, 2020-12-08:
2 is from my nodes put yours node number instead.

blairfancy, 2020-12-08:
Also check on Can logger to see if RPDO i properly transmitted.

SoumyaDeshapnde, 2020-12-08:
@blairfancy do you have an example for this?

blairfancy, 2020-12-08:
Documentation is not so bad except that it use periodic request 
`node.rpdo[4].clear()
node.rpdo[4].add_variable('Application Commands', 'Command All')
node.rpdo[4].add_variable('Application Commands', 'Command Speed')
node.rpdo[4].enabled = True

# Save new configuration (node must be in pre-operational)
node.nmt.state = 'PRE-OPERATIONAL'
node.tpdo.save()
node.rpdo.save()

# Start RPDO4 with an interval of 100 ms
node.rpdo[4]['Application Commands.Command Speed'].phys = 1000
node.rpdo[4].start(0.1)
node.nmt.state = 'OPERATIONAL'`

but if you want to send only one PDO just use:
`node.rpdo[4].transmit()` instead of `node.rpdo[4].start(0.1)`

SoumyaDeshapnde, 2020-12-08:
I'm getting a KeyError:1
My script looks like this:
```
import canopen
import time

network = canopen.Network()
network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate=125000)

node = canopen.RemoteNode(1, '/path_to_EDSFile.eds')
network.add_node(node)

# Read current PDO configuration
node.tpdo.read()
node.rpdo.read()

node.rpdo[1].clear()
node.rpdo[1].add_variable('Platform State')
node.rpdo[1].enabled = True
node.rpdo[1].transmit()

node.nmt.state = 'PRE-OPERATIONAL'
node.tpdo.save()
node.rpdo.save()

node.rpdo[1]['Platform State'].phys = 0x02
node.rpdo[1].start(1)
node.nmt.state = 'OPERATIONAL'

# Stop transmission of RxPDO
node.rpdo[1].stop()
```

```
Error:
File "CANopendemo2.py", line 17, in <module>
node.rpdo[1].clear()
KeyError: 1
```

blairfancy, 2020-12-08:
I have you checked your eds file? it must match device dictionary (read).

blairfancy, 2020-12-08:
`import canopen
import time

network = canopen.Network()
network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate=125000)

node = canopen.RemoteNode(1, '/path_to_EDSFile.eds')
network.add_node(node)

# Read current PDO configuration
node.tpdo.read()
node.rpdo.read()

node.rpdo[1].clear()
node.rpdo[1].add_variable('Platform State')
node.rpdo[1].enabled = True

node.nmt.state = 'PRE-OPERATIONAL'
node.tpdo.save()
node.rpdo.save()
node.nmt.state = 'OPERATIONAL'

node.rpdo[1]['Platform State'].phys = 0x02
node.rpdo[1].transmit()

# Stop transmission of RxPDO
node.rpdo[1].stop()`

SoumyaDeshapnde, 2020-12-08:
> I have you checked your eds file? it must match device dictionary (read).

What do you mean?

SoumyaDeshapnde, 2020-12-08:
> ```
> import canopen
> import time
> 
> network = canopen.Network()
> network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate=125000)
> 
> node = canopen.RemoteNode(1, '/path_to_EDSFile.eds')
> network.add_node(node)`
> 
> node.nmt.state = 'PRE-OPERATIONAL'
> 
> # Read current PDO configuration`
> node.tpdo.read()
> node.rpdo.read()
> 
> node.rpdo[1].clear()
> node.rpdo[1].add_variable('Platform State')
> node.rpdo[1].enabled = True
> 
> 
> node.tpdo.save()
> node.rpdo.save()
> node.nmt.state = 'OPERATIONAL'
> 
> node.rpdo[1]['Platform State'].phys = 0x02
> node.rpdo[1].transmit()
> 
> # Stop transmission of RxPDO
> node.rpdo[1].stop()
> ```

Getting the same error with this

SoumyaDeshapnde, 2020-12-08:
@blairfancy I know what is the issue. The EDS has all read-only access type variables. 
Is there a way to send a blank/custom message from a particular ID?

SoumyaDeshapnde, 2020-12-10:
Any suggestions?
I'm really stuck and need answers!

christiansandberg, 2020-12-10:
Could you upload the EDS file?

SoumyaDeshapnde, 2020-12-10:
Sure.
[EDSFile.txt](https://github.com/christiansandberg/canopen/files/5675535/EDSFile.txt)

christiansandberg, 2020-12-11:
The EDS file is missing all receive PDO entries.

SoumyaDeshapnde, 2020-12-11:
Oh! Okay I created a new one. 
[MPS_BMS_Node1_Test.txt](https://github.com/christiansandberg/canopen/files/5680234/MPS_BMS_Node1_Test.txt)

closed 2024-06-14 by friederschueler