canopen-python/canopen #210: Need exemple on how to code SDO , PDO , NMT , EMCY callback

blairfancy, 2020-12-08:
Hi,
Do you have some example of code that use callback to capture  SDO , PDO , NMT , EMCY.
We can use can listener to capture can messages but is there any way to capture object instaed?

christiansandberg, 2020-12-09:
Do you mean that you want to passively monitor a CANopen bus and decode the messages? That's not supported unfortunately.

blairfancy, 2020-12-11:
I see that callback has been implemented for PDO and EMCY messages but not for SDO and NMT messages

af-silva, 2020-12-15:
Hi @blairfancy, that is by design.
PDO and EMCY are broadcast type messages, or UDP like if you prefer, therefore the callbacks are use to listen when a new message as arrived. On the other hand SDO, are TCP like, we send the request and stay waiting (with timeout) the response arrival. NMT are internal state machine states that are handle internally by the API managing for the devices. There are other mechanism available to listen to the CANBus network, such one is the candump in linux that can be use to log the communications in realtime on the console or dump to a file for latter analysis (I have done this many time to make sure i was sending and reading the right stuff). Using the library to do this job i don't think is the best approach, but if you have a specific use case please share it for discussion and if you want you also can implement and submit a PR. @christiansandberg understands better than I the core implementation of the library. This is only my opinion. 

If i can help please drop a message/comment here.

Thanks.

ericbaril72, 2021-02-28:
Hello Christian,
When using your library, do you only use it whithin CLI scripts ? or do you use a graphical UI ?  
suggestions on UI based tools that uses you library ?

Thanks for sharing your CANopen knowledge with this lib.  I might be able to Submit some PR if I knew it would add value to the 'development Path' you envision for the library. 

CANopen can quickly become overwhelming when looking at it 'Low-Level' when using high-level production tools in work environment.  The high-level Software ( CodeSYS per example ) handles a lot of initialisation stuff and if we use your library, it most likely mean we do not have access to 'Vector' software suite.  :-)
Your library helped me a few times by comparing canbus message patterns with your code to better graps what the hell the PLC was doing.  

I do have a 'Python-Flask' Web-Server that I use with python-can /cantools that does decoding but is not  'structured' right as of now.  It does decode J1939 and CANopen but I do not always have the devices 'DBC' or 'EDS'.

- I often simulate CANopen Slave devices to test my main PLC program
- I am prototyping CANopen Slave 'Models' but not yet with your library

Let me know if CANopen Slave is something you'd use or what other tool you would recommend to emulate Device so I don't reinvent the wheel.  It might convince my to structure my project a bit and put it on Github.
Maybe your comments may show me a different approach to my 'Model' prototype that would bring an awesome PR.

let's see how it can go... throwing a message in a bottle !

friederschueler, 2024-06-14:
> Do you mean that you want to passively monitor a CANopen bus and decode the messages? That's not supported unfortunately.

It kinda is (in meantime?), as you can reuse a lot of the framework to process the raw can messages. For example I have an SDO-Logger implemented like this:

    def log_sdo(self, msg):
        try:
            if 0x580 < msg.arbitration_id <= 0x67F:
                node_id = msg.arbitration_id & 0x7F
                if node_id not in self.network.nodes:
                    logger.warning(f"SDO for Node {node_id} received but node is not in network.")
                    return
                node = self.network.nodes[node_id]
                header = msg.data[0]
                index = (msg.data[2] << 8) + msg.data[1]
                subindex = msg.data[3]
                sdo_object = node.sdo[index]
                if header in {0x2f, 0x2b, 0x27, 0x23}:  # write headers with values
                    op = "W"
                elif header in {0x4f, 0x4b, 0x47, 0x43}:  # read headers with values
                    op = "R"
                else:
                    # ignore other operations
                    return
                if isinstance(sdo_object, canopen.sdo.base.SdoVariable):
                    self.print_sdo_variable(node_id, sdo_object, op, msg.data)
                elif isinstance(sdo_object, canopen.sdo.base.SdoRecord):
                    sub_object = sdo_object[subindex]
                    self.print_sdo_variable(node_id, sub_object, op, msg.data, parent=f"['{sdo_object.od.name}']")
        except Exception as e:
            logger.error(e)
            logger.error(f"Error while receiving msg: {hex(msg.data)}")

    @staticmethod
    def print_sdo_variable(node_id, sdo_object, operation, data, parent=""):
        od = sdo_object.od
        expected_size = len(od) // 8
        data = data[4:]
        data = data[:expected_size]
        logger.info(f"SDO{operation} 0x{node_id:03x} {parent}['{od.name}']: {hex(od.decode_raw(data))}")

It is called with `network.notifier.add_listener(self.log_sdo)` and similar things could be implemented for NMT and EMCY messages. It will log all SDO Reads and Writes with Names from the Object Directory and the actual values. It will filter out the response on write messages and the request on read messages. 

To respond to @ericbaril72 there is still #47 open. We would be happy if anyone is interested in implementing the missing parts or help design them. I use python canopen myself to simulate (client) devices for testing, it works pretty well. We should close this issue if the original question was answered.

closed 2024-07-09 by friederschueler