canopen-python/canopen #211: Errors when PDO.READ()

ArthurREGNARD, 2020-12-09:
Hello,

Python version : 3.7
CANOpen version : 1.2.0

Got an error when I'm doing that:

```bash
import canopen
import os

os.system('sudo ip link set can1 type can bitrate 125000')
os.system('sudo ifconfig can1 txqueuelen 65536')
os.system('sudo ifconfig can1 up')

# Config for 2-CAN HAT
channel = "can1"
bustype = "socketcan"

# Config Node 8
node_id = 8
node_obj = "eds_files/myfile.eds"

network = canopen.Network()
network.connect(channel=channel, bustype=bustype)
node = network.add_node(node=node_id, object_dictionary=node_obj)

# Read current PDO configuration
node.tpdo.read()
node.rpdo.read()
```
```error
Traceback (most recent call last):
  File "canOpen.py", line xxx, in <module>
    node.rpdo.read()
  File "/home/pi/.local/share/virtualenvs/application-8nU-x-Et/lib/python3.7/site-packages/canopen/pdo/base.py", line 55, in read
    pdo_map.read()
  File "/home/pi/.local/share/virtualenvs/application-8nU-x-Et/lib/python3.7/site-packages/canopen/pdo/base.py", line 322, in read
    for subindex in range(1, nof_entries + 1):
TypeError: 'float' object cannot be interpreted as an integer
```

In the file `.eds` I got some reference with Varaible / Array / Record.
When I'm using SDO, I haven't got any errors. 
The error look like  Python error.

ArthurREGNARD, 2020-12-09:
For all entries of my `file.eds`, got `sub0` always with the same content :

```bash
[20AB]
ParameterName=ENABLE  MOTION
ObjectType=0x09
SubNumber=14
; Scaling=1
; Unit=OPTION

[20ABsub0]
ParameterName=NumberOfEntries
ObjectType=0x07
DataType=0x0007
DefaultValue=13
AccessType=ro

[20ABsub1]
ParameterName=ParamValue
ObjectType=0x07
DataType=0x0007
DefaultValue=0x01
LowLimit=0x00
HighLimit=0x01
AccessType=rw

[20ABsub2]
ParameterName=RESERVED
ObjectType=0x07
DataType=0x0007
DefaultValue=0x0001
AccessType=ro

[20ABsubA]
ParameterName=ExtParamName
ObjectType=0x07
DataType=0x0009
DefaultValue=ENABLE  MOTION
AccessType=ro

[20ABsubB]
ParameterName=RESERVED
ObjectType=0x07
DataType=0x0007
DefaultValue=0x18040100
AccessType=ro

[20ABsubC]
ParameterName=ParamFormat
ObjectType=0x07
DataType=0x0007
DefaultValue=0x00005500
AccessType=ro

[20ABsubD]
ParameterName=RESERVED
ObjectType=0x02
DataType=0x000F
AccessType=ro

[20ABsubE]
ParameterName=RESERVED
ObjectType=0x07
DataType=0x0007
DefaultValue=0x00000000
AccessType=ro
```

christiansandberg, 2020-12-10:
Can you post how the PDO map entries look like in the eds? Like 1400sub0.

ArthurREGNARD, 2020-12-10:
```bash
[1400]
ParameterName=Receive PDO Communication Parameter 0
ObjectType=0x9
SubNumber=3

[1400sub0]
ParameterName=Number of entries
ObjectType=0x7
DataType=0x0005
AccessType=ro
DefaultValue=2
PDOMapping=0
LowLimit=0x02
HighLimit=0x05

[1400sub1]
ParameterName=COB ID
ObjectType=0x7
DataType=0x0006
AccessType=ro
DefaultValue=$NODEID+0x200
PDOMapping=0
LowLimit=0x00000001
HighLimit=0xFFFFFFFF

[1400sub2]
ParameterName=Transmission Type
ObjectType=0x7
DataType=0x0005
AccessType=ro
DefaultValue=0x0
PDOMapping=0
```

christiansandberg, 2020-12-10:
Not sure why that would happen. You may need to trace back why that value becomes a float.

friederschueler, 2024-06-14:
Closing this now, as the problem is really old. Feel free to create a new issue with a more recent canopen version.

closed 2024-06-14 by friederschueler