canopen-python/canopen #213: Connect to two CAN Networks with Peak USB

superengineer22, 2020-12-18:
How would I connect to two CAN Networks at the same time?  I am using 2 PEAK CAN to USB adapters.

This code works great for scanning nodes on the network:
`import canopen
import time

network = canopen.Network()
network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate='1000000')

network.scanner.search()

time.sleep(0.1)
for node_id in network.scanner.nodes:
    print("Found nodes %d" % node_id)

network.disconnect()
print("Disconnected from Network")
`

My unsuccessful attempt to connect to 2 networks:
`import canopen
import time

network = canopen.Network()
network2 = canopen.Network()
network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate='1000000')  #
network2.connect(bustype='pcan', channel='PCAN_USBUS1', bitrate='1000000')

network.scanner.search()
network2.scanner.search()

time.sleep(0.1)
for node_id in network.scanner.nodes:
    print("Found nodes %d" % node_id)
for node_id in network2.scanner.nodes:
    print("Found nodes in network2 %2" % node_id)

network.disconnect()
network2.disconnect()
print("Disconnected from Networks")`

Both networks work fine separately.

christiansandberg, 2020-12-19:
I assume you pass different arguments to the connect method? In that case it is probably a limitation in the PCAN drivers.

ngblume, 2020-12-21:
I'm assuming that you got two of the single-interface adapters like https://www.peak-system.com/PCAN-USB.199.0.html

network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate='1000000') #
network2.connect(bustype='pcan', channel='PCAN_USBUS**2**', bitrate='1000000')

Every USB-CAN adapter from PEAK is listed as "PCAN_USBBUS**Z**"
If you got a pro version (https://www.peak-system.com/PCAN-USB-Pro-FD.366.0.html), this one device already will be listed as "PCAN_USBBUS1" and "PCAN_USBBUS2". 
An additional simple adapter would be "PCAN_USBBUS3"..

Hope this helps..

superengineer22, 2020-12-23:
> I'm assuming that you got two of the single-interface adapters like https://www.peak-system.com/PCAN-USB.199.0.html

Those are the adapters that I am using.

> network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate='1000000') #
> network2.connect(bustype='pcan', channel='PCAN_USBBUS**2**', bitrate='1000000')
> 
> Every USB-CAN adapter from PEAK is listed as "PCAN_USBBUS**Z**"
> If you got a pro version (https://www.peak-system.com/PCAN-USB-Pro-FD.366.0.html), this one device already will be listed as "PCAN_USBBUS1" and "PCAN_USBBUS2".
> An additional simple adapter would be "PCAN_USBBUS3"..
> 
> Hope this helps..

Unfortunately, I am out of town till the new year. I'll give this a try and update on my results when I return.
Thanks for your help!

superengineer22, 2021-01-05:
Was able to test the post from @ngblume.  It worked great!

superengineer22, 2021-01-05:
I tested the suggestion from @ngblume.  It worked great!  I added a third Peak USB card.  For reference, this code works: 

```
import canopen
import time

network = canopen.Network()
network2 = canopen.Network()
network3 = canopen.Network()

network.connect(bustype='pcan', channel='PCAN_USBBUS1', bitrate='1000000')
network2.connect(bustype='pcan', channel='PCAN_USBBUS2', bitrate='1000000')
network3.connect(bustype='pcan', channel='PCAN_USBBUS3', bitrate='1000000')

network.scanner.search()
network2.scanner.search()
network3.scanner.search()

time.sleep(0.1)
for node_id in network.scanner.nodes:
    print("Found nodes in network 1 %d" % node_id)
for node_id in network2.scanner.nodes:
    print("Found nodes in network 2 %d" % node_id)
for node_id in network3.scanner.nodes:
    print("Found nodes in network 3 %d" % node_id)

network.disconnect()
network2.disconnect()
network3.disconnect()
print("Disconnected from Networks")
```

closed 2021-01-05 by superengineer22

ngblume, 2021-01-06:
Hej hej,

Glad to be of help !
The PEAK adapters are very good, but the standard API is a little bit tricky in my opinion.

Just as a warning, that you might run into another issue:
Per adapter there can only by one custom program accessing the device aside from official PEAK software.
So running your program and some other program (maybe like a manufacturer software to configure the CANopen device) trying to access the same port like "PCAN_USBBUS3" won't work with the basic API (https://www.peak-system.com/PCAN-Basic.239.0.html?&L=1).
This requires the developer version (https://www.peak-system.com/PCAN-Developer-4.461.0.html?&L=1).
See attached picture for details...
[image]
Just keep that in mind, when you are getting weird errors.

One other tip (a little selfish):
https://github.com/ngblume/Peak-PCAN-Pied-Piper
A little program I wrote for using the PEAK adapters to forward the CAN messages to WireShark which dissects the messages (CAN and CANopen) and makes them readable for humans (https://wiki.wireshark.org/CANopen).

Cheers
Niels Göran