canopen-python/canopen #214: catch sdo exception

fabio-mat, 2020-12-22:
Hello 

I try to catch sdo exceptions and print them to my wx GUI.
Normaly this works fine, except if my canopen slave is turned off and gives no answer.
Then i get also this exeption:
```
wx._core.wxAssertionError: C++ assertion "m_count == -1 || m_count == -2" failed at ..\..\src\msw\textctrl.cpp(262) in UpdatesCountFilter::UpdatesCountFilter(): wrong initial m_updatesCount value
```

This is my function were i handle the exeptions and print them to my gui:
```
def getSdoData(self, object_name, node, print_comlog):`
        string = ""
        try:    
            data = node.sdo[object_name].raw   
        except canopen.sdo.exceptions.SdoCommunicationError as e:
            self.comlog.AppendText("SDO Communication Error: %s\n" % e)
        except canopen.sdo.exceptions.SdoAbortedError as e:
            self.comlog.AppendText("SDO Abort Error: %s\n" % e)                    
        except KeyError as e:
            self.comlog.AppendText("Key Error: %s\n" % e)    
        except canopen.objectdictionary.ObjectDictionaryError as e:
            self.comlog.AppendText("Object Dict. Error: %s\n" % e)    
        else:
            if( print_comlog == True ):
                address = self.node_fvp.object_dictionary[object_name].index 
                self.comlog.AppendText("Read parameter '%s' with adress 0x%X: %d \n" % (object_name, address, data) ) 
            string = str(data) 
            
        return(string)
```

and this is the Exception response:
```
File "C:\Python38\Lib\site-packages\canopen\sdo\client.py", line 66, in read_response
  response = self.responses.get(
File "C:\Python38\Lib\queue.py", line 178, in get
  raise Empty

_queue.Empty

During handling of the above exception, another exception occurred:

￼ File "C:\PythonWorkspace\SerAppCELDA\ServAppCELDA.py", line 1361, in getSdoData
  data = node.sdo[object_name].raw
File "C:\Python38\Lib\site-packages\canopen\variable.py", line 74, in raw
  value = self.od.decode_raw(self.data)
File "C:\Python38\Lib\site-packages\canopen\variable.py", line 36, in data
  return self.get_data()
File "C:\Python38\Lib\site-packages\canopen\sdo\base.py", line 108, in get_data
  return self.sdo_node.upload(self.od.index, self.od.subindex)
File "C:\Python38\Lib\site-packages\canopen\sdo\client.py", line 117, in upload
  fp = self.open(index, subindex, buffering=0)
File "C:\Python38\Lib\site-packages\canopen\sdo\client.py", line 197, in open
  raw_stream = ReadableStream(self, index, subindex)
File "C:\Python38\Lib\site-packages\canopen\sdo\client.py", line 243, in __init__
  response = sdo_client.request_response(request)
File "C:\Python38\Lib\site-packages\canopen\sdo\client.py", line 85, in request_response
  return self.read_response()
File "C:\Python38\Lib\site-packages\canopen\sdo\client.py", line 69, in read_response
  raise SdoCommunicationError("No SDO response received")

canopen.sdo.exceptions.SdoCommunicationError: No SDO response received

During handling of the above exception, another exception occurred:

￼ File "C:\PythonWorkspace\SerAppCELDA\ServAppCELDA.py", line 1527, in <module>
  app.MainLoop()
File "C:\Python38\Lib\site-packages\wx\core.py", line 2224, in MainLoop
  rv = wx.PyApp.MainLoop(self)
￼ File "C:\PythonWorkspace\SerAppCELDA\ServAppCELDA.py", line 1355, in data_log_read_button_pressed
  self.update_data_logger()
￼ File "C:\PythonWorkspace\SerAppCELDA\ServAppCELDA.py", line 1312, in update_data_logger
  self.list_data_log.SetItem(index, 1, self.getSdoData(object1, self.node_fvp, False) )
￼ File "C:\PythonWorkspace\SerAppCELDA\ServAppCELDA.py", line 1363, in getSdoData
  self.comlog.AppendText("SDO Communication Error: %s\n" % e)

wx._core.wxAssertionError: C++ assertion "m_count == -1 || m_count == -2" failed at ..\..\src\msw\textctrl.cpp(262) in UpdatesCountFilter::UpdatesCountFilter(): wrong initial m_updatesCount value
```

christiansandberg, 2020-12-22:
This is probably a wx question but you should probably run SDO requests in a separate thread and not on the main thread as that will lock up the application while waiting for a response. When you need to update the GUI you should do that using wx.CallAfter.

friederschueler, 2024-06-14:
Close this as it is answered without further responses.

closed 2024-06-14 by friederschueler