canopen-python/canopen #215: No SDO response received

iliya52, 2021-01-03:
Hi,
I've been trying to use canopen to test the sdo download and upload for a project I am currently working on but I keep getting the error, "No SDO response received". I've tried almost everything but I don't seem to figure this out. It would be great if anyone could help with this.

This is my code for the SDO download:
OD = os.path.join(os.path.dirname(__file__), OBJECT_DICTIONARY[0])
NODE = self.NETWORK.add_node(16, OD)
NODE.sdo.RESPONSE_TIMEOUT = 0.01
SDO_DATA_DOWNLOAD = bytearray([0x00, 0x00, 0x00, 0x00])
NODE.sdo.download(40, 0, SDO_DATA_DOWNLOAD)

Best,
Iliya

af-silva, 2021-01-05:
Hi @iliya52 ,

Take a look [here](https://canopen.readthedocs.io/en/latest/sdo.html)

The download/upload functionality is provided to write/read objects that are not provided directly in the Object Dictionary. If you have an Object Dictionary why would you not use it?

``` 
# Simple variables
device_type = node.sdo[0x1000]

# Complex records
# By name 
command_all = node.sdo['ApplicationCommands']['CommandAll']
# By object
vendor_id = node.sdo[0x1018][1].raw

```

friederschueler, 2024-06-14:
Closing this really old issue as no further reply exists.

closed 2024-06-14 by friederschueler