canopen-python/canopen #217: Reading DefaultValue from EDS is wrong for data type OCTET_STRING and DOMAIN

mlederhi, 2021-01-15:
The DefaultValue for data type OCTET_STRING and DOMAIN is formatted with two hex digits for each byte in the EDS - see corresponding CiA standard. Also the content of the default value has to end up in a bytes instance and not in a string instance. (otherwise an exception is thrown when the SDOServer tries to answer a SDO upload request for an object dictionary entry with this data type.)

Please see the following patch, and please integrate it:

--- eds.py.orig	2021-01-13 15:50:52.906669000 +0100
+++ eds.py	2021-01-15 12:48:02.089073320 +0100
@@ -138,7 +138,9 @@
 
 
 def _convert_variable(node_id, var_type, value):
-    if var_type in objectdictionary.DATA_TYPES:
+    if var_type in (objectdictionary.OCTET_STRING, objectdictionary.DOMAIN):
+        return bytes.fromhex(value)
+    elif var_type in (objectdictionary.VISIBLE_STRING, objectdictionary.UNICODE_STRING):
         return value
     elif var_type in objectdictionary.FLOAT_TYPES:
         return float(value)

christiansandberg, 2021-01-19:
Thank you for that finding. Would you be able to provide a pull request? I don't have a development environment setup right now.

mlederhi, 2021-01-20:
Created pull requerst #219.

closed 2021-01-20 by christiansandberg