canopen-python/canopen #221: Can't read hexadecimal nodeID from DCF

wvanderp, 2021-02-01:
hallo, 

While loading my dcf file i got the following error:

     source = './bin/SOMEDCF.dcf', node_id = 10
  
      def import_eds(source, node_id):
          eds = RawConfigParser()
          if hasattr(source, "read"):
              fp = source
          else:
              fp = open(source)
          try:
              # Python 3
              eds.read_file(fp)
          except AttributeError:
              # Python 2
              eds.readfp(fp)
          fp.close()
          od = objectdictionary.ObjectDictionary()
          if eds.has_section("DeviceComissioning"):
              od.bitrate = int(eds.get("DeviceComissioning", "Baudrate")) * 1000
              od.node_id = int(eds.get("DeviceComissioning", "NodeID"))
    E           ValueError: invalid literal for int() with base 10: '0x0A'
  
    ../../../.local/lib/python3.8/site-packages/canopen/objectdictionary/eds.py:36: ValueError

A part of the dcf:

    [DeviceComissioning]
    NodeID=0x0A
    NodeName=DEVICE_NAME
    BaudRate=1000
    NetNumber=1

The issue is solve by changing the nodeID to decimal but cia-306 states the following:

    Integer numbers shall be written as decimal numbers, hexadecimal numbers or octal numbers.
    Hexadecimal numbers are preceded by 0x. Octal numbers shall start with a leading 0 (not followed by
    x). If the entry contains a number the following lines are identical:
    keyname=10
    keyname=0xa
    keyname=0x0a
    keyname=0xA
    keyname=0x000A
    keyname=012
source cia 306 4.3 on page 9

So it think that all of them should be accepted.

I have implemented the workaround, so this does not have that high of a priority for me. But i thought you would like to know.

thanks in advance,

wvdp

af-silva, 2021-02-04:
Hi @wvanderp ,

Can you provide a Pull Request with the changes. It is a valid point.

Thanks.

wvanderp, 2021-02-16:
I don't have a fix for the liberary. 

i fixed it by changing the dcf I use.

commit 1b12c971 (2021-02-20): Add support for hex node ID in DCF

Fixes #221
closed 2021-02-20 by christiansandberg